OpenTabletDriver/OpenTabletDriver #1957: It is impossible to disable lock aspect ratio in 6.0, also 6.0.1 still shows as 6.0.0

Doksha, 2022-01-15:
## Description

Even when the application shows lock aspect ratio as disabled, the width and height are still linked making it impossible for the user to change the aspect ratio of the output.
(also for some reason the new release, 6.0.1 still shows as 6.0.0 in the application name and will notify you that there is a new version)
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 6.0.x ???
| Tablet                   | Veikk A15

InfinityGhost, 2022-01-16:
The version bug is known and is fixed for v0.6.0.2.

Doksha, 2022-01-17:
Should I create a different issue for the aspect ratio thing or just keep it here?

InfinityGhost, 2022-01-17:
Changing the title and editing the original issue report is enough.

gonX, 2022-01-17:
This is now the issue on aspect ratio, since we need to track that anyway. v0.6.0.2 has just been released.

gonX, 2022-01-17:
As a current workaround after disabling aspect ratio lock, you need to make sure to save your configuration and restart the GUI portion of the driver.

jamesbt365, 2022-08-27:
fixed in #2352

closed 2022-08-27 by jamesbt365