OpenTabletDriver/OpenTabletDriver #1958: UX app doesn't reflect changes made with console

sbstratos79, 2022-01-15:
## Description

UX app doesn't reflect changes made with console

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 11
| OpenTabletDriver Version | 0.6.0.1
| Tablet                   | Huion 1060 PRO

jamesbt365, 2022-01-15:
This should be considered to be added later, allowing the daemon to update to the ux the same as the ux can change the daemon.

InfinityGhost, 2022-01-18:
FWIW the UI is loosely connected to the daemon itself, its mostly a one-way bind system. It is only intended to be used by one client at a time

There are multiple issues with trying to do bind two-way that we'd have to discuss further to actually make it work however.

1. Should it synchronize all clients immediately if another one submits? A problem with this is that this would clear any settings on that client, which is undesirable. Maybe storing if any settings have been changed since would fix this, but it certainly does complicate things.
2. Should it always be synchronized? This entirely resolves the first issue but introduces a ton of complexity and potentially even more undesired effects; if one were using their tablet to change these settings it could quickly cause compounding effects when moving/resizing an area.

I'm certainly interested in improving this, however it is best done if/when we switch to [.NET MAUI](https://docs.microsoft.com/en-us/dotnet/maui/what-is-maui) therefore I'm putting this on the backlog.