OpenTabletDriver/OpenTabletDriver #1965: Daemon and client version mismatch should be more resilient

gonX, 2022-01-16:
## Description
Right now we get an exception with stuff like `No method by the name 'GetDevices' is found.' which isn't helpful for the user.

The check could probably be comprised of just having the build ID match, and if they don't match, refuse to connect and inform the user.

In theory we could probably kill the daemon and autospawn it ourselves from UX, but I'm not sure how well that translates to e.g. Linux.

InfinityGhost, 2022-01-18:
This could be as simple as adding a function that returns a `Version`, or we could make this as complex as a hash of the shared `OpenTabletDriver.Desktop` assembly. I could see potential issues with the hash method but it would be much more reliable than a version number.

gonX, 2026-09-25:
Fixed in v0.6.8 with #4834

closed 2026-09-25 by gonX