OpenTabletDriver/OpenTabletDriver #1966: Support Request: OpenTabletDriver does not work correctly.

Ltrinch, 2022-01-16:
# Support Request
On the 1st run of OpenTabletDriver I got [this error.](https://imgur.com/RaByTKQ) After that, the error went away, but OpenTabletDriver did not work correctly.
[Video with all the errors.](https://streamable.com/ty2ou5)
## What I've already tried
Tried to install libx11-dev, dotnet-sdk-6.0 libs, tried to reinstall OpenTabletDriver, but it didn't help...
I did everything from [this guide](https://opentabletdriver.net/Wiki/Install/Linux)(Ubuntu/Debian), but something went wrong...

## Diagnostic Information
I can't attach the Diagnostics because the program gives an error. (see video)

Name | Value
-- | --
Operating System | Linux Mint 20.3
OpenTabletDriver Version | 0.6.0.0
Tablet | XP-Pen G640 Rev.2

gonX, 2022-01-16:
Please check that `libxrandr-dev` is installed

Ltrinch, 2022-01-16:
Installed it, now everything works. Thank you!

closed 2022-01-16 by Ltrinch

gonX, 2022-01-16:
This is a known packaging issue and is fixed for 0.6.0.2. Sorry for the inconvenience.