OpenTabletDriver/OpenTabletDriver #1972: Filters are no longer ordered

Kuuuube, 2022-01-17:
## Description

Previously (0.6.0, 0.6.0.1) filters were listed alphabetically. On 0.6.0.2 they are now randomized like they were in <=0.5.3.3. 

I suspect this was caused by #1939.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Win10
| OpenTabletDriver Version | 0.6.0.2

Kuuuube, 2025-09-01:
Dropping this off of the 0.6.6 milestone since it's kindof a "hell issue" last time I tried to fix this. But if anyone wants to give it a go, this would be great to have in.

Mrcubix, 2025-09-07:
I don't see any traces of ordering in either the plugin manager nor in PluginSettingStoreCollection or related calls to order anything.
If anything it would be weird that that it suddenly was ordered.
(prob has more to do with the order i nwhich plugins were ordered)

Kuuuube, 2025-09-07:
It's possible it is ordered by default in the assemblies loaded. For example Circular Area is named Circular Area and all the filters also say Circular Area on the front of each filter. So if the filters are loaded in order it would make it seem like it's sorted.

Even though simply having a filter that doesn't match this naming scheme would mess it all up.

Thanks for looking into this.

gonX, 2025-09-08:
Fixed with #4049

I also had the same findings as Cubix (that it was never intentionally sorted), so I guess I should more consistently write up my findings in the future 😅

PR #4049 (2025-09-07): [0.6.X] Order Filters & Tools by Name
Closes #1972

Also fixes a bug with the selection not being set back to the previously selected plugin.
closed 2025-09-08 by gonX