OpenTabletDriver/OpenTabletDriver #1986: RPM package still not working on Fedora 35

korewaChino, 2022-01-19:
Probably is a duplicate of #1945, but even with the new dependency updates I am still unable to install the RPM version.
```
> sudo dnf install ./OpenTabletDriver.rpm 
Last metadata expiration check: 1:01:18 ago on Wed 19 Jan 2022 08:06:10 PM +07.
Error: 
 Problem: conflicting requests
  - nothing provides mono(Microsoft.Win32.Primitives) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(Microsoft.Win32.Primitives) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(Microsoft.Win32.Registry) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(Newtonsoft.Json) = 12.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Buffers) = 4.0.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections.Concurrent) = 4.0.14.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections.Concurrent) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections.Immutable) = 1.2.5.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Collections.Immutable) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.ComponentModel) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.ComponentModel.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Console) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.Debug) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.Debug) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.Process) = 4.2.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.Process) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.Tools) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.Tools) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Diagnostics.Tracing) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.IO.Compression) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.IO.Compression.ZipFile) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.IO.FileSystem) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.IO.Pipes) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.IO.Pipes) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Linq) = 4.2.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Linq) = 4.2.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Linq) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Linq) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Linq.Expressions) = 4.2.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Linq.Expressions) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Linq.Parallel) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Memory) = 4.0.1.1 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Memory) = 4.1.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Memory) = 4.2.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Memory) = 4.2.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Memory) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Net.Http) = 4.2.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Net.Http) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Net.Primitives) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Net.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Net.Sockets) = 4.2.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Net.WebSockets) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Net.WebSockets) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Numerics.Vectors) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.ObjectModel) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.ObjectModel) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Reflection.Emit) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Reflection.Emit.ILGeneration) = 4.0.3.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Reflection.Emit.ILGeneration) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Reflection.Emit.Lightweight) = 4.0.3.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Reflection.Emit.Lightweight) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Reflection.Primitives) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Reflection.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Resources.ResourceManager) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Resources.ResourceManager) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Runtime) = 4.2.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Runtime) = 4.2.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
[... 21 lines cut ...]
  - nothing provides mono(System.Text.Encoding.Extensions) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Text.Json) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Text.RegularExpressions) = 4.2.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Text.RegularExpressions) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Text.RegularExpressions) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading.Overlapped) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading.Overlapped) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading.Tasks) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading.Tasks) = 4.1.2.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading.Tasks.Dataflow) = 4.6.5.0 needed by opentabletdriver-0.6.0.2-1.x86_64
[... cut ...]
  - nothing provides mono(System.Threading.ThreadPool) = 5.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading.ThreadPool) = 6.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Threading.Timer) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(System.Xml.XDocument) = 4.1.1.0 needed by opentabletdriver-0.6.0.2-1.x86_64
  - nothing provides mono(netstandard) = 2.0.0.0 needed by opentabletdriver-0.6.0.2-1.x86_64
(try to add '--skip-broken' to skip uninstallable packages)
```

InfinityGhost, 2022-01-19:
Updated the package on the v0.6.0.2 release. It appears that the ubuntu container that the RPM builds aren't properly built, we'll likely be disabling it for future releases until we figure out a proper fix.

cindrmon, 2022-03-31:
> Updated the package on the v0.6.0.2 release. It appears that the ubuntu container that the RPM builds aren't properly built, we'll likely be disabling it for future releases until we figure out a proper fix.

Hi! Hopefully you would do the same for v0.6.0.3 and future RPM releases as well. It is still currently not working and the same errors appear when installing the latest version.

cindrmon, 2022-05-20:
Update: v0.6.0.4-1 still has the same issue. looks like it only started at v0.6.0.3.

```
-> % sudo dnf install ./OpenTabletDriver.rpm            
Last metadata expiration check: 0:07:40 ago on Fri 20 May 2022 11:38:01 PST.
Error: 
 Problem: conflicting requests
  - nothing provides mono(Microsoft.Win32.Primitives) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(Microsoft.Win32.Primitives) = 5.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(Microsoft.Win32.Registry) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(Newtonsoft.Json) = 12.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Buffers) = 4.0.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections) = 5.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections.Concurrent) = 4.0.14.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections.Concurrent) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections.Immutable) = 1.2.5.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Collections.Immutable) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.ComponentModel) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.ComponentModel.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Console) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.Debug) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.Debug) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.Process) = 4.2.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.Process) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.Tools) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.Tools) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 5.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Diagnostics.Tracing) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.IO.Compression) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.IO.Compression.ZipFile) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.IO.FileSystem) = 5.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.IO.Pipes) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.IO.Pipes) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Linq) = 4.2.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Linq) = 4.2.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Linq) = 5.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Linq) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Linq.Expressions) = 4.2.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Linq.Expressions) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Linq.Parallel) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Memory) = 4.0.1.1 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Memory) = 4.1.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Memory) = 4.2.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Memory) = 4.2.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Memory) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Net.Http) = 4.2.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Net.Http) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Net.Primitives) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Net.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Net.Sockets) = 4.2.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Net.WebSockets) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Net.WebSockets) = 4.1.2.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Numerics.Vectors) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
[... cut ...]
  - nothing provides mono(System.Threading.ThreadPool) = 5.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Threading.ThreadPool) = 6.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Threading.Timer) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(System.Xml.XDocument) = 4.1.1.0 needed by opentabletdriver-0.6.0.4-1.x86_64
  - nothing provides mono(netstandard) = 2.0.0.0 needed by opentabletdriver-0.6.0.4-1.x86_64
(try to add '--skip-broken' to skip uninstallable packages)
```

hwsmm, 2022-05-20:
@cindrmon Try my package [here](https://github.com/hwsmm/OpenTabletDriver.Packaging/releases/download/test/OpenTabletDriver.rpm)

OP(@korewaChino) and I tried fixing it but our PR didn't get merged for months

I just don't want to tell Infinity to do something about this anymore :P

cindrmon, 2022-05-21:
> @cindrmon Try my package [here](https://github.com/hwsmm/OpenTabletDriver.Packaging/releases/download/test/OpenTabletDriver.rpm)
> 
> OP(@korewaChino) and I tried fixing it but our PR didn't get merged for months
> 
> I just don't want to tell Infinity to do something about this anymore :P

Ohh ok, thanks for the link, it installed flawlessly! By the way, would you know if [this repo](https://github.com/hwsmm/OpenTabletDriver.Packaging/) you got the rpm package from would be maintained for future releases? Thank you.

jamesbt365, 2022-05-21:
Theres a pr out into the main repo for this package if you check with this change into it.

thatpix3l, 2022-07-20:
For those that arrive to this thread by searching online and absolutely _need_ the latest version, here's what to do (at least on Fedora):

1. Download version [0.6.0.2](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.6.0.2) of the software, and `dnf install` the rpm. This step is to just get all of the dependencies, that are actually available in Fedora's repos.
2. Go to the [latest](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/latest) tag, download the portable version called `OpenTabletDriver.linux-x64.tar.gz` and decompress it
3. In the compressed folder, directly run the daemon binary in `./OpenTabletDriver/OpenTabletDriver.Daemon`
4. If you need to configure stuff, directly run the GTK GUI binary in `./OpenTabletDriver/OpenTabletDriver.UX.Gtk`
5. ???
6. Profit

jamesbt365, 2022-07-20:
Or you could use hwsmm rpm linked above as they are the maintainer for that.

willswats, 2022-09-06:
Can't get the driver to work on Fedora Silverblue which @hwsmm provided, and it was working when I was using Fedora Workstation. Anyone know how I can get it to work on Silverblue?

jamesbt365, 2022-09-06:
Use the release tarballs and put the binaries in your path

nmoseman, 2023-05-24:
I installed the package from: 
 https://github.com/hwsmm/OpenTabletDriver.Packaging/releases/download/test/OpenTabletDriver.rpm

And it worked for me in Fedora 38.  Running Wayland Gnome. 

Had to rrmod the hid_uclogic driver.  I had it blacklisted, I think it is loaded with the initrc. Not sure. 

This is with a Huion Kamvas 13 pro 2.5k. 

I previously tried the RPM from this Copr: https://copr.fedorainfracloud.org/coprs/cappyishihara/opentabletdriver/.  That installed, but it was the 6.0.3 release which didn't have the 13 pro 2.5k support enabled yet.

On a side note:

I have noticed that this video tablet is very sensitive to the USB-C cable.  I have video-out through USB-C on my laptop and the level of success depended heavily on the cable.   

Typically the display is detected and comes up, but it errors out detecting the HID devices. Swapping cables caused it to work.  It is very consistent.

So if somebody is having problems try the HDMI  3-way cable first before trying a single combined USB-C.

glow12121, 2023-06-14:
I have Gaomon S620 and the 6.0.4 release says no tablet detected. Is the tablet not compatible yet? It works fine under the native Gaomon driver (which I want to avoid). Help needed.

*Fedora 38 with Gnome*

X9VoiD, 2023-08-13:
Already fixed

closed 2023-08-13 by X9VoiD