OpenTabletDriver/OpenTabletDriver #1987: Regression from 0.5.3.3: No reports on FT-0405-U

xax, 2022-01-19:
## Description
The (elderly) Wacom FT-0405-U tablet ist recognized as such in releases 0.6.0.x, but the pointer does not move on using the pen.

The *Tablet Debugger* is showing plausible *Raw Tablet Data*, but *Tablet Report* remains blank. In Version 0.5.3.3, on the other hand, the TF-0405-U works and *Tablet Report* shows plausible interpreted data (`ReportID`, `Position`, `Pressure`, `PenButtons`).

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10 v21H2 b19044.1466
| OpenTabletDriver Version | 0.6.0.0, 0.6.0.1. 0.6.0.2
| Tablet                   | Wacom FT-0405-U

jamesbt365, 2022-01-19:
Can you exprot a diagnostic please? Help > export diagnostics, save that file as a txt and send it here.

xax, 2022-01-19:
`"Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'."` with v0.5.3.3 versus `"Using report parser type 'OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser'"` with v0.6.0.2 catches my eye…

• [Wacom-FT-0405-U-0.5.3.3.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7899481/Wacom-FT-0405-U-0.5.3.3.json.txt)
• [Wacom-FT-0405-U-0.6.0.2.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7899485/Wacom-FT-0405-U-0.6.0.2.json.txt)

xax, 2022-01-19:
Byte 1 (second byte) of *Raw Tablet Data* as shown by *Tablet Debugger* v0.6.0.2 is `0x90` . I mention that with regards to https://github.com/OpenTabletDriver/OpenTabletDriver/blob/502589660232add52a1ae7bd7de584e7192f2ec2/OpenTabletDriver.Configurations/Parsers/Wacom/Intuos/IntuosReportParser.cs#L18

gonX, 2022-01-19:
This is likely caused by https://github.com/OpenTabletDriver/OpenTabletDriver/pull/1698 - if you end up making a PR try not to break that change.

There's a chance checking for either bit is fine

jamesbt365, 2022-02-28:
Try the bamboo parser.

jamesbt365, 2022-06-18:
as i suspected, solved by changing the parser in #2246

closed 2022-06-18 by jamesbt365