OpenTabletDriver/OpenTabletDriver #1988: Lock to usable area, Clamp input outside area and Ignore input outside area do not function

Kuuuube, 2022-01-19:
## Description

Lock to usable area, Clamp input outside area and Ignore input outside area all get stuck to default and cannot be changed.

Reproduction steps
1. Right click the area editor
2. Change Lock to usable area, Clamp input outside area or Ignore input outside area to their non-default state

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Win10
| OpenTabletDriver Version | 0.6.0.2

gonX, 2022-01-20:
Probably not directly related to the issue, but to lock aspect ratio the user has to save the config and restart OTD entirely. Does that workaround work here?

jamesbt365, 2022-01-20:
From what i saw in discord and what i tested myself this functionality just gets stuck and cannot be changed.

Kuuuube, 2022-01-21:
From further testing it seems that only Ignore input outside area is affected. Instead of stopping input when outside the tablet area it appears to clamp or send the cursor to the bottom right instead of ignore.

I have appended the issue with this information and updated the reproduction steps to be more obvious.

InfinityGhost, 2022-02-01:
Moving to v0.6.1 due to merging of #2035

PR #2352 (2022-08-21): Dependency Injection Rewrite
## Notes

- Completely rewrote the OpenTabletDriver Daemon and UI projects from scratch using Dependency Injection.
- Merged the OpenTabletDriver.Plugin library into the core OpenTabletDriver library.

## Issues

- Fixes #1988
closed 2022-08-21 by InfinityGhost