OpenTabletDriver/OpenTabletDriver #1999: Hawku and Devocub filters feel weird after minimizing OTD into the taskbar

Gadzzaa, 2022-01-23:
## Description
Whenever i minimize OTD into the taskbar the filters start feeling 60hz on polling rate instead of the value i set in the plugin settings, same happens when running the daemon instead of the UX and sadly i cannot try this on windows 10 neither linux since i own a 12th gen intel cpu. Nothing shows up in the console and when testing on osu it shows the right polling rate but it still feels like it is set to 60hz or lower.

The only temporary fix i've found is to re-enable the filters and save&apply while not minimizing the OTD and letting the UX run in the background.

Do you guys know what might be causing this? Is it maybe a windows update causing this? It started happening a few weeks ago for me and until then it was fine.

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 11 |
| OpenTabletDriver Version | v0.6.0.0 |
| Tablet                   | CTL-472 |

Sublimelime, 2022-01-25:
Does this happen if you start OTD minimized? To do this, create a shortcut to the OpenTabletDriver.UX.Wpf.exe executable, then go to the properties of the shortcut and look for a dropdown you can change to "Minimized" (can't remember the tab it's under as I'm not on a windows machine right now). Then, launch OTD through that shortcut.

Alternatively, if you're familiar with the command line, invoke the executable with the `-m` flag.

We've had very strange issues with people on windows 11, so I want to see if windows is doing something when you click minimize (lowering process priority?) vs starting it minimized.

Gadzzaa, 2022-01-26:
This also happens when i start OTD minimized

iceflarexd, 2022-05-06:
i know this doesnt add much to the discussion but i just updated to windows 11 and im getting the same behavior, i decided to look for a bug report of it already before making my own

xxCherry, 2022-06-26:
As said on MSDN: 
[image]
[MSDN link](https://docs.microsoft.com/en-us/windows/win32/api/timeapi/nf-timeapi-timebeginperiod) (remarks)

So the following code will be needed to fix this.
```csharp
PowerThrottlingState state = new();

state.Version = 1;
state.ControlMask = PROCESS_POWER_THROTTLING_IGNORE_TIMER_RESOLUTION;
state.StateMask = 0;

SetProcessInformation(GetCurrentProcess(), 
                      ProcessPowerThrottling,
                      state, 
                      sizeof(state));
```
Also you might need to call `NtSetTimerResolution`.

After I did the above, the problem was fixed.

X9VoiD, 2022-12-31:
Fixed by #2534

closed 2022-12-31 by X9VoiD