OpenTabletDriver/OpenTabletDriver #2000: "Reports recorded" in tablet debugger should be hidden while not recording

gonX, 2022-01-24:
## Description
This isn't exactly intuitive despite the title of the textbox.

Setting `Visible = false` in code doesn't reflow the UI, so the fix is less simple than expected.

If reflowing the UI is hard we could probably drop this text box entirely from the UI. We can also make a fixed-width spinner (alternating `/ - \ |`) instead so that the user at least knows something is being logged.

Sublimelime, 2022-01-25:
I disagree with just hiding it - once you've stopped recording, it's potentially useful to know how many reported you record**ed**, past tense.

Instead, how about hiding it until some reports are recorded, then displaying it (permanently after)? Though tbh this is kinda a non-issue as the text box is pretty small and keeping it as-is doesn't strike me as problematic.

gonX, 2022-01-25:
Yes, that's what I meant. We should only show it if we have recorded in this session. Thanks for the correction.

It has confused some users before thinking that reports aren't being parsed correctly, which is why I want the change

gonX, 2025-01-24:
Moving milestone to 0.6.6 because Avalonia branch has this considered.

gonX, 2025-09-08:
Fixed in #4048

closed 2025-09-08 by gonX