OpenTabletDriver/OpenTabletDriver #2005: Daemon Exited Object Reference not set to an instance of an object

TheFantasticLoki, 2022-01-24:
## Description

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |
| OpenTabletDriver Version |
| Tablet                   |

jamesbt365, 2022-01-24:
Fill this out and collect the stack.

TheFantasticLoki, 2022-01-24:
did not mean to hit enter and submit

TheFantasticLoki, 2022-01-24:
@Jamesbt365 Updated with log

jamesbt365, 2022-01-24:
- close otd
- Delete %localappdata%\OpenTabletDriver\settings.json or move them out the folder.

closed 2022-01-24 by jamesbt365

TheFantasticLoki, 2022-01-24:
OTD Still crashes attaching new log below:

1/24/2022 4:42:48 PM
System.NullReferenceException
Object reference not set to an instance of an object.
OpenTabletDriver.Desktop
   at OpenTabletDriver.Desktop.Profiles.AreaSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.AbsoluteModeSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.Profile.GetDefaults(TabletReference tablet)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection.<>c.<.ctor>b__2_0(TabletReference s)
   at System.Linq.Utilities.<>c__DisplayClass2_0`3.<CombineSelectors>b__0(TSource x)
   at System.Linq.Enumerable.SelectIListIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Collections.ObjectModel.ObservableCollection`1.CreateCopy(IEnumerable`1 collection, String paramName)
   at System.Collections.ObjectModel.ObservableCollection`1..ctor(IEnumerable`1 collection)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 profiles)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 tablets)
   at OpenTabletDriver.Desktop.Settings.GetDefaultProfiles()
   at OpenTabletDriver.Desktop.Settings.GetDefaults()
   at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings()
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
GetDefaults

jamesbt365, 2022-01-24:
make a folder called `userdata` by the exes of otd.

TheFantasticLoki, 2022-01-24:
That worked.