OpenTabletDriver/OpenTabletDriver #2013: UX Crash when resizing tablet area

joshuah345, 2022-01-26:
## Description

OpenTabletDriver UX crashes when the tablet area is invalid (eg width/height empty or zero)

Seems like it tries to draw the rectangle (area) instead of displaying the "invalid area" warning

[Link to log](https://termbin.com/0alz)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System    |macOS 12.0.1 Monterey     
| OpenTabletDriver Version |0.6.0.2
| Tablet                |  Wacom CTH-680

jamesbt365, 2022-01-27:
Pretty sure the warning on safe should be sufficient to say that it will cause issues 
[image]

joshuah345, 2022-01-27:
i didn't press save. it aborts when emptying the value

jamesbt365, 2022-09-04:
fixed in #2352

gonX, 2025-01-24:
Fixed in #2248 / v0.6.5

PR #2248 (2024-11-24): Add missing error messages
Fix #2013
closed 2025-01-24 by gonX