OpenTabletDriver/OpenTabletDriver #2014: Tip Pressure breaks Artist Mode

Nominup, 2022-01-27:
## Description

Changing the tip pressure while on artist mode prevents me from using the tablet while on artist mode, as if it doesn't detect my tablet (while it says it does and the tablet debugger works fine), this keeps being the case even when turning artist mode on and off, redetecting my tablet or even restarting open tablet driver, requiring a restart of my computer to fix.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System  |  Ubuntu 18.04   |
| OpenTabletDriver  Version | 0.5.3.1 |
| Tablet |   Wacom CTL-672  |

jamesbt365, 2022-01-27:
You are meant to unbind the tip for artist mode otherwise it crashes.

Nominup, 2022-01-27:
Oh, it should really say that somewhere. It seems to automatically do that the first time too, without displaying it has, making it even more confusing.

closed 2022-01-27 by Nominup

jamesbt365, 2022-01-27:
Doesn't crash in the later releases.