OpenTabletDriver/OpenTabletDriver #2024: Support Request: OTD not working without Root and cursor is jumpy

akiiinee, 2022-01-28:
# Support Request

When I use OTD, i have to run it using Sudo because launching it normally from the terminal doesn't work (no change in tablet area).
When OTD is running as Root, the cursor keeps snapping from the left to the right side of the screen, when its still in one place.

## What I've already tried
Reinstall, Reboot, Different Tablet (Current one is Wacom PTZ-930, other tested one is Wacom CTL-480)

## Diagnostic Information

```
{
  "App Version": "OpenTabletDriver v0.6.0.2+35df25a",
  "Build Date": "26/01/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.16.2.1",
    "VersionString": "Unix 5.16.2.1"
  },
  "Environment Variables": {
    "USER": "root",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "/home/asuka/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/cuda/bin:/opt/cuda/nsight_compute:/opt/cuda/nsight_systems/bin:/home/asuka/.dotnet/tools:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-5/1-5:1.1/0003:046D:C335.0004/hidraw/hidraw3",
      "Manufacturer": "Logitech",
      "ProductName": "Gaming Keyboard G910",
      "SerialNumber": "0D69385E3934",
      "FriendlyName": "Gaming Keyboard G910",
      "VendorID": 1133,
      "ProductID": 49973,
      "InputReportLength": 64,
      "OutputReportLength": 64,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.0/0003:056A:00B2.0001/hidraw/hidraw0",
      "Manufacturer": "Tablet",
      "ProductName": "PTZ-930",
      "SerialNumber": "",
      "FriendlyName": "PTZ-930",
      "VendorID": 1386,
      "ProductID": 178,
      "InputReportLength": 10,
      "OutputReportLength": 0,
      "FeatureReportLength": 9,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2:1.3/0003:08BB:2902.0002/hidraw/hidraw1",
      "Manufacturer": "Burr-Brown from TI              ",
      "ProductName": "USB Audio CODEC ",
      "SerialNumber": "",
      "FriendlyName": "USB Audio CODEC ",
      "VendorID": 2235,
      "ProductID": 10498,
      "InputReportLength": 2,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
[... 1547 lines cut ...]
      "Group": "DeviceIOException",
      "Message": "No serial number.",
      "StackTrace": "   at HidSharp.Platform.Linux.LinuxHidDevice.GetSerialNumber()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d)\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2022-01-28T22:58:38.3436117+01:00",
      "Group": "IOException",
      "Message": "Read-only file system : '/hdd/Akari/monkey'",
      "StackTrace": "   at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirectory, Func`2 errorRewriter)\n   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String path, OpenFlags flags, Int32 mode)\n   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)\n   at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)\n   at System.IO.Strategies.FileStreamHelpers.ChooseStrategy(FileStream fileStream, String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, Int64 preallocationSize)\n   at System.IO.FileInfo.OpenWrite()\n   at OpenTabletDriver.UX.MainForm.ExportDiagnostics()",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2022-01-28T22:58:38.3436304+01:00",
      "Group": "IOException",
      "Message": "Read-only file system : '/hdd/Akari/monkey'",
      "StackTrace": "   at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirectory, Func`2 errorRewriter)\n   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String path, OpenFlags flags, Int32 mode)\n   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)\n   at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)\n   at System.IO.Strategies.FileStreamHelpers.ChooseStrategy(FileStream fileStream, String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, Int64 preallocationSize)\n   at System.IO.FileInfo.OpenWrite()\n   at OpenTabletDriver.UX.MainForm.ExportDiagnostics()",
      "Level": 3,
      "Notification": false
    }
  ]
}
```

vedattt, 2022-01-28:
It seems you have a Wacom driver installed already. Uninstall it and try again.

jamesbt365, 2022-01-28:
Do not run otd as root.

akiiinee, 2022-01-28:
If i run without root, both tablets do not interact with the driver at all, literally nothing happens

jamesbt365, 2022-01-28:
Well thats where you read the console output or go to help > export diagnostics to actually figure out the problem instead of masking with root. OTD is not meant to be run as root.

akiiinee, 2022-01-28:
> It seems you have a Wacom driver installed already. Uninstall it and try again.

I'm running a relatively fresh Arch Linux install and only installed OTD 2 days ago

akiiinee, 2022-01-28:
> Well thats where you read the console output or go to help > export diagnostics to actually figure out the problem instead of masking with root. OTD is not meant to be run as root.

That's what I did, and it didn't show anything. The console was exactly the same output with and without root.

jamesbt365, 2022-01-28:
you are blindly ignoring what i said now.

Sublimelime, 2022-01-29:
Hey there @akiiinee, sorry for the confusion - the issue is the built-in wacom kernel module (which provides some basic functionality to wacom tablets) is interfering with OTD. You can see the error in the diagnostic,

> "Message": "Detected input coming from Wacom driver",

To fix, run the commands from our FAQ [here](https://opentabletdriver.net/Wiki/FAQ/Linux#ghost-cursor), 

```
echo "blacklist wacom" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod wacom
```

and afterwards reboot. You should be able to use OTD **without having to run it as root** from there. Let us know if things still don't work.

akiiinee, 2022-01-29:
Thank you @Sublimelime  for the the fix. Disabling the wacom module fixed the jumping however it still required launching using sudo, as it wouldn't do anything running as normal user again.

jamesbt365, 2022-01-29:
Thats because you are running the session as root which you shouldn't do and is a big hole in security.

closed 2022-01-29 by jamesbt365

akiiinee, 2022-01-29:
Sorry what? I'm currently running as a standard user (see attachment). OTD does not work for me if its not run as root, which is odd.

[image]

jamesbt365, 2022-01-29:
The session, not user.

akiiinee, 2022-01-29:
Running OTD without root after the fix also results in this crash
[image]

jamesbt365, 2022-01-29:
Pipe is stuck because you ran it as root, reboot.

jamesbt365, 2022-01-29:
same error because i ran it as root.
[image]

akiiinee, 2022-01-29:
Video of what happens when run without root right after rebooting

https://user-images.githubusercontent.com/38049865/151665640-92393a45-8411-4983-8634-62e05eee7c9a.mp4

jamesbt365, 2022-01-29:
Thats because when you load otd as root it uses the root settings in /root/.config/OpenTabletDriver instead of ~/.config/OpenTabletDriver, What you are seeing here is normal.