OpenTabletDriver/OpenTabletDriver #2025: Support Request: OTD cannot detect Huion HS64 tablet on Manjaro Linux

JamTheDev, 2022-01-29:
# Support Request
I need help! My OpenTabletDriver is not detecting my HuionHS64 tablet. I am using Manjaro (KDE Plasma) Linux. 
I installed OpenTabletDriver (0.6.2-2) from AUR using pikaur.

## What I've already tried
https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1304

## Diagnostic Information

Here is my [diagnostics file](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7962571/diagnostic.txt)

If theres anything ive missed, please let me know!

jamesbt365, 2022-01-29:
https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception
https://opentabletdriver.net/Wiki/FAQ/Linux#fail-device-streams

JamTheDev, 2022-01-29:
I have finally got it working, thank you so much!

closed 2022-01-29 by JamTheDev