OpenTabletDriver/OpenTabletDriver #2031: Wacom CTE-440 It does not work.

gitasau, 2022-01-30:
## Description
It is recognized, but the cursor and pen do not respond.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |Windows10
| OpenTabletDriver Version |0.6.0.2
| Tablet                   |Wacom CTE-440

jamesbt365, 2022-01-30:
This is a support request not a bug, go to help > export diagnostics, save that file as a txt and send it here.

gitasau, 2022-01-30:
[CTE-440_output.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7967048/CTE-440_output.json.txt)
Send the output file.

jamesbt365, 2022-01-30:
Make a folder called `Configurations` inside the folder that the otd executables are in and put this file in, don't forget to remove the txt at the end of the file name leaving it just as `CTE-440.json` [CTE-440.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7967053/CTE-440.json.txt)

gitasau, 2022-01-30:
It worked properly. Thank you very much.

closed 2022-01-30 by jamesbt365