OpenTabletDriver/OpenTabletDriver #2033: DeviceUnauthorizedAccessException

krypton1101, 2022-01-31:
## Description
Console output:
[Detect:Info]	Searching for tablets...
[Detect:Warning]	More than 1 matching device has been found.
[Detect:Info]	Found tablet 'Huion HS611'
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Device:Info]	Device disconnected.
[Detect:Info]	Searching for tablets...
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw8.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw9.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw8.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw9.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw8.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw9.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw8.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw9.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw8.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw9.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw8.
[... 461 lines cut ...]
[Huion HS611:Info]	Display area: [1920x1080@<2880, 540>:0°],
[Huion HS611:Info]	Tablet area: [258.4x161.5@<129.2, 80.75>:0°],
[Huion HS611:Info]	Clipping: Enabled
[Huion HS611:Info]	Ignoring reports outside area: Disabled
[Huion HS611:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Huion HS611:Info]	Pen Bindings: Mouse Button Binding: Middle, Mouse Button Binding: Right
[Settings:Info]	Driver is enabled.
[Huion HS611:Info]	Output mode: Artist Mode
[Huion HS611:Info]	Display area: [1920x1080@<2880, 540>:0°],
[Huion HS611:Info]	Tablet area: [258.4x161.5@<129.2, 80.75>:0°],
[Huion HS611:Info]	Clipping: Enabled
[Huion HS611:Info]	Ignoring reports outside area: Disabled
[Huion HS611:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Huion HS611:Info]	Pen Bindings: Mouse Button Binding: Middle, Mouse Button Binding: Right
[Settings:Info]	Driver is enabled.

Driver works at all, but pen buttons don`t work with any mouse buttons binded to them (if bind to pen buttons smth else, like key binding or multi-key binding it works). And there are some errors in console.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Ubuntu LTS20.04
| OpenTabletDriver Version | v.0.6.0.2
| Tablet                   | Huion HS611

jamesbt365, 2022-01-31:
Well, help > export diagnostics, save that file as a txt and add it as an attachment.

jamesbt365, 2022-01-31:
also https://opentabletdriver.net/Wiki/FAQ/Linux#fail-device-streams

krypton1101, 2022-01-31:
If it matter, when starting throw console, it prints (OpenTabletDriver.UX.Gtk:48447): Gtk-WARNING **: 22:56:59.126: __gtksharp_1_Eto_GtkSharp_Forms_EtoVBox 0x55b5c25369f0 adjusted size vertical min 25 natural 760 must not decrease below min 25 natural 2415
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7973841/diagnostics.txt)

krypton1101, 2022-01-31:
If not secret, how to restart them?)

jamesbt365, 2022-01-31:
`systemctl --user restart opentabletdriver`

krypton1101, 2022-01-31:
Ok, no more errors.
[Detect:Info]	Searching for tablets...
[Detect:Warning]	More than 1 matching device has been found.
[Detect:Info]	Found tablet 'Huion HS611'
[Huion HS611:Info]	Output mode: Artist Mode
[Huion HS611:Info]	Display area: [1920x1080@<2880, 540>:0°],
[Huion HS611:Info]	Tablet area: [258.4x161.5@<129.2, 80.75>:0°],
[Huion HS611:Info]	Clipping: Enabled
[Huion HS611:Info]	Ignoring reports outside area: Disabled
[Huion HS611:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Huion HS611:Info]	Pen Bindings: OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding, OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding
[Settings:Info]	Driver is enabled.
[Device:Info]	Device disconnected.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Detect:Info]	Searching for tablets...
[Detect:Warning]	More than 1 matching device has been found.
[Detect:Info]	Found tablet 'Huion HS611'
[Huion HS611:Info]	Output mode: Artist Mode
[Huion HS611:Info]	Display area: [1920x1080@<2880, 540>:0°],
[Huion HS611:Info]	Tablet area: [258.4x161.5@<129.2, 80.75>:0°],
[Huion HS611:Info]	Clipping: Enabled
[Huion HS611:Info]	Ignoring reports outside area: Disabled
[Huion HS611:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Huion HS611:Info]	Pen Bindings: OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding, OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding
[Settings:Info]	Driver is enabled.
..But binding mouse buttons still don`t work.

jamesbt365, 2022-01-31:
Try: https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception

krypton1101, 2022-01-31:
krypton11@krypton11-desktop:~$ echo "blacklist hid_uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
[sudo] password for krypton11:
blacklist hid_uclogic
krypton11@krypton11-desktop:~$ sudo rmmod hid_uclogic
rmmod: ERROR: Module hid_uclogic is not currently loaded

InfinityGhost, 2022-08-26:
Really old issue at this point, reopen if this is still a problem.

closed 2022-08-26 by InfinityGhost