OpenTabletDriver/OpenTabletDriver #2034: Sorry once again Crashing : System.InvalidOperationException: Sequence contains no elements

TheFantasticLoki, 2022-02-01:
## Description

Crashing once again at start Kernelbase.dll exception
Here is the only crash info I could gather:

Application: OpenTabletDriver.Installer.exe
CoreCLR Version: 4.700.20.36602
.NET Core Version: 3.1.7
Description: The process was terminated due to an unhandled exception.
Exception Info: System.InvalidOperationException: Sequence contains no elements
   at System.Linq.ThrowHelper.ThrowNoElementsException()
   at System.Linq.Enumerable.First[TSource](IEnumerable`1 source)
   at InstallerLib.Downloader.GetLatestRelease(Repository repo)
   at InstallerLib.Installer.CheckForUpdate()
   at OpenTabletDriver.Installer.MainForm.UpdateControls(Boolean autostart)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__139_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at System.Windows.Threading.DispatcherOperation.InvokeInSecurityContext(Object state)
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.Run()
   at System.Windows.Application.RunDispatcher(Object ignore)
   at System.Windows.Application.RunInternal(Window window)
   at System.Windows.Application.Run(Window window)
   at Eto.Wpf.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.Installer.Wpf.MainClass.Main(String[] args)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 Pro |
| OpenTabletDriver Version | 0.6.0.2 |
| Tablet | XP-PEN Star 06C |

jamesbt365, 2022-02-01:
Don't use the installer, use the portable releases instead. We don't support the installer.

closed 2022-02-01 by jamesbt365

TheFantasticLoki, 2022-02-01:
Launching UX.wpf does not launch daemon

sensejui79, 2022-02-01:
version: 2
updates:
  - package-ecosystem: gradle
    directory: "/"
    schedule:
      interval: "daily"
    open-pull-requests-limit: 3

gonX, 2022-02-01:
> Launching UX.wpf does not launch daemon

Make sure they're both located in the same folder

sensejui79, 2022-02-01:
> [#2013 ](url)
[introduction-to-github (1).pdf](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7978969/introduction-to-github.1.pdf)
[shaktiio-brochure.pdf](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7978973/shaktiio-brochure.pdf)

TheFantasticLoki, 2022-02-01:
> > Launching UX.wpf does not launch daemon
> 
> Make sure they're both located in the same folder

They are both in the same bin folder which also contains another userdata folder.

jamesbt365, 2022-02-01:
delete the internals of the userdata folder.