OpenTabletDriver/OpenTabletDriver #2038: Plugin idea: mappable "soft buttons"

Boolean263, 2022-02-02:
## Description
It would be useful to be able to register touches on the digitizer that fall outside of the screen-mapped portion and read them as extra "auxiliary" buttons with definable features.

## Inspiration 
The UC-Logic WP8060 tablet (repackaged by several vendors) has button outlines printed along the edges of its digitizer area. The vendor driver can map touches in each of these outlined areas to separate functions. (It can also disable this feature.)

Attached picture is from a google image search, hopefully it demonstrates the idea. The dark gray area is the extent of the digitizer; the printed grid is where the vendor driver maps the screen when the soft buttons are enabled.

[image]

## UI Implementation Ideas
I understand that plugins can't change the GUI representation of the digitizer area.

The plugin could have tickboxes to enable/disable this feature on each edge, an input for the width (for left/right edges) or height (for top/bottom edges) of the new area, and an input for the number of buttons into which to divide the area.

It would be up to the user to ensure that they left sufficient space beyond the screen-mapped area on their digitizer for these soft buttons to exist.

Mrcubix, 2022-02-02:
If i understand what you want, you basically want to map unused part of the tablet area as aux keys of various size

Boolean263, 2022-02-02:
That's exactly it, yes.