OpenTabletDriver/OpenTabletDriver #2039: Add support for Huion H580X

ThatGuyChens, 2022-02-02:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Huion
| Product Name    | H580X
| Width           | 203.2mm
| Height          | 127mm
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information

[hel.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7988244/hel.txt)

jamesbt365, 2022-02-02:
Can you go to tablet > device string reader, put in 9580 in the top box and 100 in the middle, then click dump all.

ThatGuyChens, 2022-02-02:
I did 2 dumps incase but this is what I got from them, dump all barely functioned for some reason
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989163/dump.txt)
[dump2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989164/dump2.txt)

Mrcubix, 2022-02-02:
Do a final dump just so we can determine if it's a Gaomon 830 (RNG Edition) V2 or not

ThatGuyChens, 2022-02-02:
This is all I keep getting.

[dump7.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989285/dump7.txt)
[dump6.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989286/dump6.txt)
[dump5.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989287/dump5.txt)
[dump4.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989288/dump4.txt)

jamesbt365, 2022-02-02:
Your tablet is a bit random with the strings which is not normal but we can work with it.
remove the `.txt` from this file name and then put it in /usr/share/OpenTabletDriver/Configurations (you will have to make this folder) then go to tablet > detect tablet, it may take a few tries.
[H580X.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989342/H580X.json.txt)

ThatGuyChens, 2022-02-02:
It works! Thanks a lot.

jamesbt365, 2022-02-02:
not yet. go to Tablet > tablet debugger

jamesbt365, 2022-02-02:
Get the max x and y by moving your pen to the bottom right of your tablet, the key is to go slowly as i require as accurate as possible.

ThatGuyChens, 2022-02-02:
the width and height need to be readjusted, it's from 0, 0 to 35246, 25400

jamesbt365, 2022-02-02:
Can you check the x axis again, i think you could get higher.

ThatGuyChens, 2022-02-02:
40640 I pulled it all the way

jamesbt365, 2022-02-02:
Can you confirm the max pressure in the debugger is 8191?

And how many tablet buttons and pen buttons does this have?

ThatGuyChens, 2022-02-02:
max pressure indeed went to 8191, the tablet has 8 buttons and the pen 2

jamesbt365, 2022-02-02:
Replace the old config with this one, remember to rename it.
[H580X.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7989489/H580X.json.txt)
Then tablet > detect tablet again. Once it detects go into the debugger and check if tilt responds correctly.

ThatGuyChens, 2022-02-02:
tilt appears to be working

jamesbt365, 2022-02-02:
Can you confirm when you bind the buttons in the bindings tab that they work.

ThatGuyChens, 2022-02-02:
both pen and tablet buttons work, but auxiliary buttons show 4 extra in tablet debugger

STnetwork, 2022-02-10:
Hi, I tried to add the json configuration H580X , but the software doesn't detect the tablet, i'm on Windows.

Could you help me please ?

jamesbt365, 2022-06-06:
#2040