OpenTabletDriver/OpenTabletDriver #2050: OTD.UX can crash from file dialogs when daemon is disconnected

vedattt, 2022-02-06:
## Description

OTD.UX is crashable by opening a file dialog, exiting the daemon, and using the file dialog.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 6.0.3
| Tablet                   | Any

gonX, 2025-01-24:
If I had to guess this is probably caused by grabbing the file path before doing the daemon operation. Setting milestone for 0.6.5.1 because it's probably an easy fix - if not, we'll just push it back.

gonX, 2025-01-25:
The "Daemon closing" window prevents me from interacting with the filedialog, so I'm going to pretend this is fixed.

closed 2025-01-25 by gonX