OpenTabletDriver/OpenTabletDriver #2054: Support Request: OTD does not detect CTE-650

Ari24-cb24, 2022-02-07:
# Support Request
Hey, I installed OTD today to switch over from my old win 7 wacom driver (I use win 10). As I connected my wacom CTE-650, the tablet wasn't found by OTD.
I noticed a little message saying "Another tablet driver found: Wacom" and so I uninstalled the Wacom drivers, uninstalling every file coming from that old wacom driver as well. But the problem still occurs. (Before uninstalling the old Wacom driver, I tried using the disablewacomdrivers.bat file from TabletDriverGUI but that didn't work as well)
  
I tested this with OTD 5.3.3, OTD 5.3 and OTD 6.0.3 but none of them worked.

## What I've already tried
https://opentabletdriver.net/Wiki/FAQ/Windows#uclogic-supported-undetected

## Diagnostic Information
https://pastebin.com/fjhzGQxF

Ari24-cb24, 2022-02-07:
Both, first uninstalled it via windows and then deleted the restover

Ari24-cb24, 2022-02-08:
So I checked some things and it seems like the uninstaller actually didnt uninstall everything. I may have to remove the old driver manually so this doesnt really seem like an issue from odt.
Will keep this issue updated.

Mrcubix, 2022-02-08:
Right click on start menu > device manager > view > device by container > find your tablet there > open it's container (arrow on the left) > Screenshot the container

Ari24-cb24, 2022-02-08:
[image]
Its on german so heres the translation
- HID-Compliant Mouse
- HID-conform pen
- HID conform digitizing device
- HID conform, device defined by the manufacturer
- Wacom Tablet

Mrcubix, 2022-02-08:
Wacom drivers were indeed still installed, right click on `Wacom Tablet` > Update > update manually (second option) > let me pick from a list > Screenshot what you have there

Ari24-cb24, 2022-02-08:
[image]
- [x] show compatible hardware

USB-Inputdevice
Wacom Tablet

_The driver has got a digital signature_

Mrcubix, 2022-02-08:
first one

Ari24-cb24, 2022-02-08:
Worked perfectly for me, thank you for spending the time helping me, really appreciate it!

Ari24-cb24, 2022-02-08:
Okay, please nevermind my comment above, it seems like the cursor does not respond when using my tablet even though the tablet debugger receives inputs from my tablet.

Here's the diagnostic: https://pastebin.com/0pa5ccMm

OTD does in fact register the tablet fine.
_Also, sorry for reopening an issue, don't know if I should've created a new one_

Ari24-cb24, 2022-02-08:
Yes, I do in fact have vanguard open.
I did not run OTD and resetted everything as well was closing the vanguard but the problem still exists.

Mrcubix, 2022-02-08:
check for `vg.exe` in task manager

Ari24-cb24, 2022-02-08:
There's no such process currently running. Also tried using `taskkill` but the process couldn't be found

Ari24-cb24, 2022-02-09:
I tried uninstalling the hardware (thats what windows said) with the drivers via device manager and reconnecting the hardware but the problem still exists.

Could this be an issue with OTD itself? OTD does receive input data from the tablet via the tablet debugger and it registers the tablet. But I can also see a message saying "Another tablet driver found: Wacom".

jamesbt365, 2022-06-18:
should be moved to its own issue if this is still occuring though was likely to do with valorant

closed 2022-06-18 by jamesbt365

Ari24-cb24, 2022-06-18:
Just a little side note, I switched to a new pc last month and it worked fine again, even with Valorant and that stuff. I did not download the wacom drivers again. I also assume that my pc was just bloated with some other drivers and maybe issues occured or stuff. If anyone encounters the same issue, a fresh installation of windows can be helpful. Anyways, thanks for helping tho!