OpenTabletDriver/OpenTabletDriver #2067: H430P Not detected

Stormer3, 2022-02-13:
Huion Tablet H430P is on the support list but is not being detected by the application

OS: Windows 10 LTSC 
.Net 6.0 installed
Tried with manufacture driver removed and installed neither resulted in detection (reboot was performed after removal of manufacture driver) 

Device manager states the huion tablet and pen are present

    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_01#7&f121d0a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet",
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },

    {
      "Time": "2022-02-13T05:02:57.3815018-05:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Huion H430P'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },****

adryzz, 2022-02-13:
Please follow the template and don't chop the diagnostics

Stormer3, 2022-02-13:
Updated using the template (hope its formatted right) and I didn't see somewhere I could attach a file. So heres the ugly paste. Btw the response time is insane thanks for the quick replies

jamesbt365, 2022-02-13:
just name it diag.txt and drag and drop it, don't paste it

Stormer3, 2022-02-13:
Updated thanks

jamesbt365, 2022-02-13:
Try [this](https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0) while i check the rest

Stormer3, 2022-02-13:
That did the trick actually. I suppose uninstalling the software itself isn't enough. Even went into device manager and uninstalled the devices there and then reconnected the device.

Stormer3, 2022-02-13:
Thanks so much!