OpenTabletDriver/OpenTabletDriver #2072: Gaomon S620 detecting input but not moving the cursor.

DarkGears, 2022-02-16:
## Description
When I open the debugger, it can see where the pen is at.  But when I use the pen, the output with the cursor isn't working.
## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System         |  Win11/OS
| OpenTabletDriver Version |  0.6.0.3
| Tablet                   |  Gaomon S620

jamesbt365, 2022-02-16:
https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

zephsalman, 2022-02-16:
hey im having the same issue, i did the windows cleanup you said and it started working then stopped again after i opened osu

jamesbt365, 2022-02-16:
As in it doesn't work specifically in osu, then just turn off raw input and don't run it as admin.

jamesbt365, 2022-02-16:
Do you have Valorant or Vanguard open?

zephsalman, 2022-02-16:
omg no way, wtf valorant

DarkGears, 2022-02-16:
im trying to use pen on valorant and i already tried using the cleanup

jamesbt365, 2022-02-16:
Use 0.5.3.3 otd and follow this.

How To Play With Valorant:
- Install and use VMulti plugin. Make sure to follow its setup instructions.
- Use VMulti Relative Mode.
- Run Valorant by tapping with your tablet (This is important since Vanguard only allows one source for "mouse" input)
- Never move your mouse while Valorant is still loading.
- Enjoy'nt ￼

Vanguard Quirks:
- As VMulti is considered a separate input source from an actual physical mouse, Vanguard imposes several limitations to it.
1. You won't be able to press left click while pressing a non-modifier keyboard key (for ex. WASD) and shift, we will call these keys from here onward as protected keys.
2. Every time you press a protected key, the next two left clicks will always be dropped.
3. If left click is held first and then a protected key is pressed, the left click will sometimes be dropped.

Please be noted that there is no fix to this, as even Logitech G Hub, Razer Synapse, HawkuTD, DevocubTD, and even XP-Pen, Huion, Gaomon, Veikk drivers are affected without exceptions. As long as the input source is not a real physical mouse, Vanguard will impose such quirks intentionally.

Vanguard will impose such quirks intentionally. - Listen especially to this bit. This isn't a bug but instead caused by Riot and there is nothing we can do about it.

closed 2022-02-16 by jamesbt365

DarkGears, 2022-02-16:
thank you so much