OpenTabletDriver/OpenTabletDriver #2074: Click not registering

k355l3r-5yndr0m3, 2022-02-16:
## Description

I recently bought a Vinsa 1060 Plus, which from what I can tell is just a rebranded 10moon 1060N. Everything works fine except that when I press the pen down, nothing happens. I opened the tablet debugger and the driver does receive the pressure data but it doesn't seem to be translating it to a left click. 
Sometime this would affect my mouse, and I cannot left click with it. Restarting Xorg enable me to left click with my mouse again but the problem  with the tablet is still present.  
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Arch linux |
| OpenTabletDriver Version | 0.6.0.2-2 |
| Tablet                   | Vinsa 1060 Plus/ 10moon 1060N |

jamesbt365, 2022-02-16:
Is his in artist mode?

k355l3r-5yndr0m3, 2022-02-16:
> Is this in artist mode?
In Artist mode, it seems be hit or miss. Sometime time the click register, sometime it doesn't. In krita, only the start of a stroke is drawn. 
To click I need to press extra hard, and for some reason my tablet stops sending input when that happends.

ds1605, 2022-03-06:
I can confirm that this issue also present for me, for the same tablet. Tapping only works in artist mode, and not any other mode. Some of the brushes in Krita only draw when some parameters are of (size variation), which leads me to think that pressure sensitivity is not fully functional.

OS: OpenSUSE Tumbleweed
Driver version: 0.6.0.3

BuMiTa, 2022-03-23:
I have the same issue with the same tablet. Is it fixed yet?

jamesbt365, 2022-03-23:
give details, this tablet is a waste of money being a scam on its own and none of the devs own one to check this out ourselves though possibly a pressure parsing error.

Operating system?
Output mode?
What has you tried?
Can you get a recording with the debugger open with you pressing the pen tip?

BuMiTa, 2022-03-23:
This tablet work quite well with it's official driver, however the official driver sucks.
OS: Windows 10 21h2
Output mode: Absolute mode
I've tried changing to another output mode.
https://user-images.githubusercontent.com/102239517/159729292-d252e03d-2bea-4983-9cb9-30644c410b4c.mp4

jamesbt365, 2022-03-23:
so pressure sits between 100 and 200 with spikes to extremely high numbers. This information helps figure this out, let us check something

BuMiTa, 2022-03-24:
> so pressure sits between 100 and 200 with spikes to extremely high numbers. This information helps figure this out, let us check something

Sorry for the late reply, but when the pressure sits between 100 and 200 is when I didn't touch the tip at all. Is it supposed to be 0?

spacexmoon, 2022-03-30:
> Is this in artist mode?

yes

emircanerkul, 2023-01-22:
Same issue. I debugged the driver and the pressure is well detected only tip detection fails.

[image]

emircanerkul, 2023-01-22:
[some debug data](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10474421/tablet-data.txt)

user202729, 2023-07-27:
Sounds like the driver actually works well, you just need to configure the tip activation threshold.

The way the calculation work is follows: Take the raw pressure value (displayed in the debugger), divided by the MaxPressure value (currently 8191, but will be changed to 1800 after #2853), multiply by 100, then if it's greater than the "Tip Activation Threshold" value then send a click.

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/5cd15773faa05533b8e536899de2efafcf7e2fc5/OpenTabletDriver.Desktop/Binding/ThresholdBindingState.cs#L25

Either way, (before the pull request) try setting Tip Activation Threshold to 5, which corresponds to an absolute threshold value of approximately 400. Also assign "Tip Button" to the left mouse click.

matous-volf, 2023-12-28:
This works, thanks!

jamesbt365, 2025-04-08:
As the support for this tablet was yanked and is no longer supported, this is no longer an issue.

closed 2025-04-08 by jamesbt365