OpenTabletDriver/OpenTabletDriver #2075: Tablet not working on fullscreen applications when using wayland

kaanyalova, 2022-02-16:
## Description

When I try to play osu-stable using wine and set the game to fullscreen i cannot move the cursor. I can move the cursor if I exit fullscreen , play the game when using xorg or use the default linux drivers .

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Manjaro Linux
| OpenTabletDriver Version | 0.6.0.2-2
| Tablet                   | Wacom ctl 490

kaanyalova, 2023-11-12:
Setting it to artist mode lets me move the cursor, I am closing this because I am no longer interested. Reopen if needed.

jamesbt365, 2023-11-12:
Giving as there should be a better way, this should not be closed.

gonX, 2025-01-24:
Closing as Absolute Mode **on Linux** should now be considered deprecated. Use Artist Mode instead. See #3496.

closed 2025-01-24 by gonX