OpenTabletDriver/OpenTabletDriver #2076: Support Request: Hovering with CTH-480 causes cursor to jump around

Exiledx, 2022-02-16:
When hovering with my pen the cursor freaks out and jumps all around. When pushed/dragged on the tablet it works fine. Any idea what might be causing this problem?

jamesbt365, 2022-02-16:
Possibly other drivers, go to help > export diagnostics, save that file with the `txt` extension and send it here.

Exiledx, 2022-02-16:
[helpgit.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8083606/helpgit.txt)

jamesbt365, 2022-02-16:
"Console Log": [
    {
      "Time": "2022-02-16T16:13:34.5197013-05:00",
      "Group": "Detect",
      "Message": "Another tablet driver found: Wacom",
      "StackTrace": null,
      "Level": 2,
      "Notification": false
    },

Exiledx, 2022-02-16:
I double checked and it should be uninstalled? Does it go by another name

jamesbt365, 2022-02-28:
Either its installed or this is just chatter, a thing in physics. Get a video of it if possible.

jamesbt365, 2022-04-06:
Closing due to no response, if you can supply what i have said above go ahead and do so and i'll reopen it.

closed 2022-04-06 by jamesbt365