OpenTabletDriver/OpenTabletDriver #2078: Mapping mouse buttons to pen doesn't work.

PolyCatGamedev, 2022-02-18:
## Description
When I go in the "Pen Settings" it lets me bind my mouse inputs to the pen buttons. So, I map my buttons, switch to "Artist Mode", hit "Apply" and "Save" and then when I press the pen button while hovering nothing happens. It works perfectly with keyboard bindings, only the mouse bindings don't work. Also the bug seems to go away when I switch to absolute mode. 
btw. I got the opentabletdriver-git package from the AUR.

## System Information:
|Name: PolyCat | Value |
| ---------------- | ----- |
| Operating System: Endeavour OS (Arch Linux)|
| OpenTabletDriver Version 0.6.0.3 |
| Tablet Huion HS610 |

darkingmimo, 2025-08-06:
the problem still persists. I mapped my pen binding, went into artist mode. Nothing. They only work for absolute mode.

yanwenywan, 2026-10-19:
I'm encountering the same issue whilst trying to map the touch bar on a huion H1161.

patching in #4804 fixed my use case and I can now use the touch bar to scroll like a mouse