OpenTabletDriver/OpenTabletDriver #2082: Performance issues with the drivers

m-ngrt, 2022-02-19:
## Description

The driver overlaps itself every time I turn my computer back on. I see what you see in the image below and if I force close it the cursor starts moving by itself until I unplug the tablet
[image]

## System Information:

| Name Martin | Value |
| ---------------- | ----- |
| Operating System Windows 10 |
| OpenTabletDriver Version | 0.5.3.3
| Tablet Wacom CTL-472|

jamesbt365, 2022-02-19:
this seems like a way that you have make it startup on boot, should be nothing to do with us.

Mrcubix, 2022-02-19:
can't replicate in 0.5.3.3, most likely an user error

jamesbt365, 2022-02-28:
This is how you start otd on boot, what you have done is user error because otd doesn't start on boot unless you make it, this below is the way that it will work. Whatever method you have done before isn't behaving, (its probably force starting it whenever it wakes up from sleep as well)

## Start OTD minimized on boot of Windows
- Right click OpenTabletDriver.UX.Wpf.exe -> Create shortcut
- Right click shortcut -> Properties -> Run -> Minimized
- Press Win + R -> Type shell:startup -> Press Enter
- Move the shortcut to the folder that opened

closed 2022-02-28 by jamesbt365