OpenTabletDriver/OpenTabletDriver #2083: [Linux] Artist Mode output configuration is misleading when on KDE Plasma Wayland session

NyaomiDEV, 2022-02-20:
## Description
On Plasma Wayland session, all tablets are supposed to be configured via the `kcm_tablet` module in System Settings (Input Method -> Drawing Tablet). This settings module is presented like this:
[image]

As you can see, OpenTabletDriver's Artist mode output is seen as a tablet by this module (which technically isn't wrong); therefore, KWin is instructed to apply whichever configuration is put here on top of the settings provided by OTD.

Now this has some problems:
- OTD's display output is incorrectly mapped: the entire display output range in OTD (= possibly multiple displays, so wonky aspect ratio) is now mapped (and stretched/fit) to whichever screen is considered by KWin as the selected one for the exposed virtual tablet. This is noticeable for people with more than one displays, or a pen display.
- Tablet orientation and left-handed mode are overridden by these settings; though probably if nobody touches these here, the ones on OTD may continue to work kind of reliably (I haven't tested those).

This is a serious issue! Since Plasma 5.24, the KDE developers ship Plasma Wayland as the default session in KDE neon. Also, the next Ubuntu releases are supposed to land with Wayland sessions as their defaults, so it's reasonable to think that Kubuntu will also ship Wayland as default; and that may very well be the same for Fedora KDE.

In practice this means a lot of current and new Plasma users could be experiencing this issue if they happen to use OTD in their systems.

## What can be done about it
If OTD is run under Plasma Wayland, a notice should warn users about this behaviour, so that they can optimally configure their devices. Also, tablet orientation (rotation) and left handed mode should be greyed out, as their configuration should ideally be left to the Wayland server.
In the long run it might be possible to directly communicate with KWin to get and set which screen is assigned to the virtual OTD tablet, so that the configuration can once again reliably be made within OTD. (Hint: D-Bus calls)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Artix Linux on Plasma Wayland |
| OpenTabletDriver Version | commit 632b0a8f |
| Tablet                   | Huion Kamvas Pro 16 |

jamesbt365, 2022-02-20:
This is a problem with the way that Plasma Wayland handles the tablet and can't exactly be fixed on our end. This is an issue with the way that plasma wayland handles this as gnome wayland handles it completely fine. So this should be reported up to plasma devs. Though something will be documented on the website.

closed 2022-02-20 by jamesbt365

yump, 2022-07-25:
To complete the link loop, this has been reported as [KDE bug 450874](https://bugs.kde.org/show_bug.cgi?id=450874).