OpenTabletDriver/OpenTabletDriver #2087: UC-Logic PF1209 not detected

mkmori, 2022-02-20:
# Support Request
UC-Logic PF1209 (Monoprice) not detected....
Win10 Home, OTD v0.6.0.3

## What I've already tried
Uninstalled all tablet drivers, (incl. Wacom).
Different USB ports.
Disabled Windows Ink via regedit.

## Diagnostic Information
```
    {
      "DevicePath": "\\\\?\\hid#vid_5543&pid_0042&col03#6&4068c3b&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "               ",
      "ProductName": "TabletPF1209\u0014",
      "SerialNumber": "",
      "FriendlyName": "TabletPF1209\u0014",
      "VendorID": 21827,
      "ProductID": 66,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_5543&pid_0042&col02#6&4068c3b&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "               ",
      "ProductName": "TabletPF1209\u0014",
      "SerialNumber": "",
      "FriendlyName": "TabletPF1209\u0014",
      "VendorID": 21827,
      "ProductID": 66,
      "InputReportLength": 6,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_5543&pid_0042&col01#6&4068c3b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "               ",
      "ProductName": "TabletPF1209\u0014",
      "SerialNumber": "",
      "FriendlyName": "TabletPF1209\u0014",
      "VendorID": 21827,
      "ProductID": 66,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_5543&pid_0042&col04#6&4068c3b&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "               ",
      "ProductName": "TabletPF1209\u0014",
      "SerialNumber": "",
      "FriendlyName": "TabletPF1209\u0014",
      "VendorID": 21827,
      "ProductID": 66,
      "InputReportLength": 0,
      "OutputReportLength": 0,
      "FeatureReportLength": 4,
      "CanOpen": true
    }
```

mkmori, 2022-02-20:
### String Dump
```
Index 1:                
Index 2: TabletPF1209
Index 3: F308PG
Index 4: Tablet Mouse
Index 5: PenKeyboad
Index 6: HIDMouse
Index 7: F308PG
Index 8: F308PG
Index 9: F308PG
Index 10: F308PG
Index 11: F308PG
Index 12: F308PG
Index 13: F308PG
Index 14: F308PG
Index 15: F308PG
Index 16: F308PG
Index 17: F308PG
Index 18: F308PG
Index 19: F308PG
Index 20: F308PG
Index 21: F308PG
Index 22: F308PG
Index 23: F308PG
Index 24: F308PG
Index 25: F308PG
Index 26: F308PG
Index 27: F308PG
Index 28: F308PG
Index 29: F308PG
Index 30: F308PG
Index 31: F308PG
Index 32: F308PG
Index 33: F308PG
Index 34: F308PG
Index 35: F308PG
Index 36: F308PG
Index 37: F308PG
Index 38: F308PG
Index 39: F308PG
Index 40: F308PG
Index 41: F308PG
Index 42: F308PG
Index 43: F308PG
Index 44: F308PG
Index 45: F308PG
Index 46: F308PG
Index 47: F308PG
Index 48: F308PG
Index 49: F308PG
Index 50: F308PG
Index 51: F308PG
Index 52: F308PG
Index 53: F308PG
Index 54: F308PG
Index 55: F308PG
Index 56: F308PG
Index 57: F308PG
Index 58: F308PG
Index 59: F308PG
Index 60: F308PG
Index 61: F308PG
Index 62: F308PG
Index 63: F308PG
Index 64: F308PG
Index 65: F308PG
Index 66: F308PG
Index 67: F308PG
Index 68: F308PG
Index 69: F308PG
Index 70: F308PG
Index 71: F308PG
Index 72: F308PG
Index 73: F308PG
Index 74: F308PG
Index 75: F308PG
Index 76: F308PG
Index 77: F308PG
Index 78: F308PG
Index 79: F308PG
Index 80: F308PG
Index 81: F308PG
Index 82: F308PG
Index 83: F308PG
Index 84: F308PG
Index 85: F308PG
Index 86: F308PG
Index 87: F308PG
Index 88: F308PG
Index 89: F308PG
Index 90: F308PG
Index 91: F308PG
Index 92: F308PG
Index 93: F308PG
Index 94: F308PG
Index 95: F308PG
Index 96: F308PG
Index 97: F308PG
Index 98: F308PG
Index 99: F308PG
Index 100: 巀䙐Ͽߐ
Index 101: 愀
Index 102: UC-Logic Technology Corp.
Index 103: F308PG
Index 104: F308PG
Index 105: F308PG
Index 106: F308PG
Index 107: F308PG
Index 108: F308PG
Index 109: F308PG
Index 110: 0
Index 111: F308PG
Index 112: F308PG
Index 113: F308PG
Index 114: F308PG
Index 115: F308PG
Index 116: F308PG
Index 117: F308PG
Index 118: F308PG
Index 119: F308PG
Index 120: F308PG
Index 121: F308PG
Index 122: F308PG
Index 123: F308PG
Index 124: F308PG
Index 125: F308PG
Index 126: F308PG
Index 127: F308PG
Index 128: F308PG
Index 129: F308PG
Index 130: F308PG
Index 131: F308PG
Index 132: F308PG
Index 133: F308PG
Index 134: F308PG
Index 135: F308PG
Index 136: F308PG
Index 137: F308PG
Index 138: F308PG
Index 139: F308PG
Index 140: F308PG
Index 141: F308PG
Index 142: F308PG
Index 143: F308PG
Index 144: F308PG
Index 145: F308PG
Index 146: F308PG
Index 147: F308PG
Index 148: F308PG
Index 149: F308PG
Index 150: F308PG
Index 151: F308PG
Index 152: F308PG
Index 153: F308PG
Index 154: F308PG
Index 155: F308PG
Index 156: F308PG
Index 157: F308PG
Index 158: F308PG
Index 159: F308PG
Index 160: F308PG
Index 161: F308PG
Index 162: F308PG
Index 163: F308PG
Index 164: F308PG
Index 165: F308PG
Index 166: F308PG
Index 167: F308PG
Index 168: F308PG
Index 169: F308PG
Index 170: F308PG
Index 171: F308PG
Index 172: F308PG
Index 173: F308PG
Index 174: F308PG
Index 175: F308PG
Index 176: F308PG
Index 177: F308PG
Index 178: F308PG
Index 179: F308PG
Index 180: F308PG
Index 181: F308PG
Index 182: F308PG
Index 183: F308PG
Index 184: F308PG
Index 185: F308PG
Index 186: F308PG
Index 187: F308PG
Index 188: F308PG
Index 189: F308PG
Index 190: F308PG
Index 191: F308PG
Index 192: F308PG
Index 193: F308PG
Index 194: F308PG
Index 195: F308PG
Index 196: F308PG
Index 197: F308PG
Index 198: F308PG
Index 199: F308PG
Index 200: F308PG
Index 201: F308PG
Index 202: F308PG
Index 203: F308PG
Index 204: F308PG
Index 205: F308PG
Index 206: F308PG
Index 207: F308PG
Index 208: F308PG
Index 209: F308PG
Index 210: F308PG
Index 211: F308PG
Index 212: F308PG
Index 213: F308PG
Index 214: F308PG
Index 215: F308PG
Index 216: F308PG
Index 217: F308PG
Index 218: F308PG
Index 219: F308PG
Index 220: F308PG
Index 221: F308PG
Index 222: F308PG
Index 223: F308PG
Index 224: F308PG
Index 225: F308PG
Index 226: F308PG
Index 227: F308PG
Index 228: F308PG
Index 229: F308PG
Index 230: F308PG
Index 231: F308PG
Index 232: F308PG
Index 233: F308PG
Index 234: F308PG
Index 235: F308PG
Index 236: F308PG
Index 237: F308PG
Index 238: F308PG
Index 239: F308PG
Index 240: F308PG
Index 241: F308PG
Index 242: F308PG
Index 243: F308PG
Index 244: F308PG
Index 245: F308PG
Index 246: F308PG
Index 247: F308PG
Index 248: F308PG
Index 249: F308PG
Index 250: F308PG
Index 251: F308PG
Index 252: F308PG
Index 253: F308PG
Index 254: F308PG
Index 255: F308PG
```

jamesbt365, 2022-02-20:
You didn't mention platform. I'm going to assume windows.

- download and install [this](https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe) tool

- When done, head to C:\Program Files\USBPcap in the explorer (you can probably copy paste that path in the path bar)

- Once you're there, start USBPcapCMD.exe

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to C:\Program  Files\USBPcap,

- Start USBPcapCMD.exe,

- screenshot the console.

mkmori, 2022-02-21:
Hi: Apologies if I'm missing something, but I didn't get a y/n prompt--it went straight to the console. Anyway, I did reboot and snip the console below. (I also got a pcap for item "3", but I guess I need Wireshark or something to read it...?)

[image]

jamesbt365, 2022-02-21:
**With manufacturer drivers running** 

1) press 3 then press enter,

2) enter a name such as capture then press enter,

3) Do the following steps:

- Move for 5s
- Unplug the tablet from your computer (remove the usb cable from the port),
- Wait 2 seconds,
- Plug in your tablet again (connect your tablet to your pc ),
- Wait 2 seconds,
- Move for 5s
- Close the console (red cross),
- A new file has been generated with the name you gave in the same folder,
- Upload that file here

mkmori, 2022-02-21:
[usbpcap3_0829.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8110847/usbpcap3_0829.zip)

InfinityGhost, 2022-08-26:
Issue is stale, reopen if the problem still exists.

closed 2022-08-26 by InfinityGhost

odisseus, 2024-06-08:
I too have this problem. Windows 10 22H2, OTD v 0.6.4.0.