OpenTabletDriver/OpenTabletDriver #2088: XP-Pen Deco Fun XS not being detected

Axis4s, 2022-02-21:
help

jamesbt365, 2022-02-21:
Fill out the template.

jamesbt365, 2022-02-21:
top of otd there is `help > export diagnostics` save that file with the `txt` file extension and send that file here.

Axis4s, 2022-02-21:
> top of otd there is `help > export diagnostics` save that file with the `txt` file extension and send that file here.
[file.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8112546/file.txt)

jamesbt365, 2022-02-21:
try out the config from #2085

jamesbt365, 2022-02-21:
also you may need [this](http://opentabletdriver.net/Wiki/FAQ/Linux#fail-device-streams)

Axis4s, 2022-02-21:
> try out the config from #2085
> 
> Make a `Configurations` folder in `/usr/share/OpenTabletDriver/` move that config file there.

that's a link to a channel, not to a file

jamesbt365, 2022-02-21:
[image]

jamesbt365, 2022-02-21:
[image]
[image]
then press `control + s`

jamesbt365, 2022-02-21:
fixed in [discord](https://discord.com/channels/615607687467761684/723399565780189234/945466793437319260)

closed 2022-02-21 by jamesbt365