OpenTabletDriver/OpenTabletDriver #2090: Support Request: OTD not detecting Huion Kamvas Pro 24

PhirosWolf, 2022-02-22:
# Support Request
I recently received my Kamvas Pro 24, and since I'm on linux, drivers for tablets are a real pain.
The issue I have is that the device is not detected by OTD.

My laptop is running linux, Manjaro (with KDE env), with kernel 5.16.7-1

I'm fully aware that my Kamvas Pro 24 is not (yet) supported by OTD, but I've tried everything I could and OTD seems to be the most up to date project with the quickest support.

Is it possible to still support this tablet (Kamvas Pro 24 from Huion) even if it is not listed in the list of supported tablets?

I'm available on Discord/Telegram if needed for quicker support (please contact me via `iow08obt3@mozmail.com` to give you my Discord/Telegram id).

## What I've already tried
- ["Device is currently in use" Wiki entry](https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception)
- "The modules reload after rebooting?" Wiki entry
- ["Failed to open device streams" Wiki entry](https://opentabletdriver.net/Wiki/FAQ/Linux#fail-device-streams)
- Blacklisting drivers for modprobe (blacklisted `wacom` and `hid_uclogic`)
- Digimend drivers
- Creating a user-space driver with evdev
- Tinkering with xinput and xsetwacom (they do not recognize correctly the sliders and some buttons are missing)
- Repluging the tablet (multiple times)
- Rebooting my laptop (multiple times)

## Diagnostic Information
- [diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8119484/diagnostics.txt)
- [getstring.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8119485/getstring.txt) (result of command `for i in {1..255}; do echo '['$i'] '`otd getstring 9580 110 $i` >> getstring.txt; done;`, blank lines are errors)
- [console.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8119490/console.txt) (I am aware that it says "Another tablet driver found: UC Logic", but I can't do anything about it, I check 3 times that it was really blacklisted for modprobe)

If diagnostic information is missing, please let me know what I should add.

Sublimelime, 2022-02-22:
Hey there,

Since you have a discord, you can join [our server](https://discord.com/invite/9bcMaPkVAR) and ask in one of the #help channels (the linux one preferably since you're on linux), due to the back-and-forth needed to support a new tablet (taking configs and trying them out), it's easier.

jamesbt365, 2022-02-28:
#2092

PR #2092 (2022-02-23): Add support for Huion Kamvas Pro 24
Verification: https://canary.discord.com/channels/615607687467761684/945769415310467092/945782595059339265

#1852 would be useful for this tablet.

> fixes #2090
closed 2022-02-28 by jamesbt365