OpenTabletDriver/OpenTabletDriver #2091: Support Request: The application crashes around 5 seconds of opening it.

Aventei, 2022-02-22:
# Support Request

I double click the application, it opens and says Connecting to OpenTabletDriver Daemon. Then in 5 seconds it crashes.
## What I've already tried

Restarting my PC.
## Diagnostic Information

jamesbt365, 2022-02-22:
Well i'm going to go off of what you have said, also you didn't use the template so i'm also going to assume you are  using Windows.

Make a `userdata` folder with the executable.

jamesbt365, 2022-02-28:
Assuming 6 days with no response that using the `userdata` folder solved this. Alternatively removing `%localappdata%\OpenTabletDriver\settings.json` and if that doesn't work then remove the `Plugins` folder as well.

If  neither of these works then i assume that you have it in a zip folder still or the daemon isn't in the same folder as the ux or the names have been tampered with.

closed 2022-02-28 by jamesbt365