OpenTabletDriver/OpenTabletDriver #2094: Add support for Huion HS611 tablet

TheDreamingOwl, 2022-02-25:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    |
| Product Name    |
| Width           |
| Height          |
| Pressure Levels |
| LPI             |

## Diagnostic Information

jamesbt365, 2022-02-25:
Can you supply a diagnostic as the template requests please, go to `help > export diagnostic` and save that file with the `txt` file extension then upload it here.

jamesbt365, 2022-02-25:
turns out i didn't check the supported list either. This is already supported and has been for a while.
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/OpenTabletDriver.Configurations/Configurations/Huion/HS611.json

closed 2022-02-25 by jamesbt365