OpenTabletDriver/OpenTabletDriver #2099: bug with tablet

NatteTosti69, 2022-02-27:
## Description

| Name             | Value |
| ---------------- | ----- |
| Operating System         |
| OpenTabletDriver Version |
| Tablet                   |

jamesbt365, 2022-02-27:
Fill it out and explain.

jamesbt365, 2022-02-27:
Your video is private, if this is only osu then turn off raw input, and if that doesn't solve it then `help > export diagnostic` save that file with the `txt` extension and upload that here.

NatteTosti69, 2022-02-27:
[file.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8148976/file.txt)

NatteTosti69, 2022-02-27:
https://www.youtube.com/watch?v=ajIQYRCExxw
its now public

jamesbt365, 2022-02-27:
I see no other other than you have your display area letterboxed, depending on if you want to do that or not. You might have wacom drivers installed however i don't see hard evidence of this, otherwise file > reset to defaults and apply then see then.

NatteTosti69, 2022-02-27:
resetting to default fixed the bug. thank you very much

closed 2022-02-27 by NatteTosti69