OpenTabletDriver/OpenTabletDriver #2106: OTD refuses to connect to the daemon once the terminal has been disowned and closed.

jamesbt365, 2022-03-04:
## Description

When disowning the daemon from the terminal and closing the terminal if the ux is opened after it will refuse to connect to the daemon again even though the daemon functions as normal. See This video example.

https://user-images.githubusercontent.com/22662856/156778529-99b4e7f7-2761-496c-933b-405090833bac.mp4

This doesn't seem to apply with distro or version of OpenTabletDriver however i have noted them down anyway. At the same time this just could be intended linux behavior when something is disowned?
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Gentoo Linux
| OpenTabletDriver Version | 0.6.0.3
| Tablet                   | N/A

gonX, 2023-11-21:
Related to #1855 probably

gonX, 2026-10-02:
This still happens on pre-0.6.8 d133af73

gonX, 2026-10-02:
Actually, to be clear, the daemon just straight up disappears if you disown it and close the terminal.

closed 2026-10-03 by gonX