OpenTabletDriver/OpenTabletDriver #2107: Add 'artist mode

fredemmott, 2022-03-04:
Is there an alternative API/output mode that applications could be modified to use to get data from OpenTabletDriver, without going via windows ink or mouse emulation?

In my case, I'm looking for an alternative to huion's wintab drivers that I could add to https://github.com/fredemmott/openkneeboard ; for this use case, it would be an added bonus if background access were supported - currently it uses DLL injection and SetWindowLongPtrA for wintab when another application is active.

From a brief look, it seems that this is possible to add, but would need a new plugin to be written first to add this functionality?

_Originally posted by @fredemmott in https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1230#issuecomment-1059217948_

fredemmott, 2022-03-04:
This would ideally also be adoptable by open source software like Krita, which currently support wintab and windows ink

jamesbt365, 2022-03-04:
I'm not quite sure how this is different from winink or wintab, can you elaborate a bit?

X9VoiD, 2022-03-04:
You may create an output mode plugin that sends data to your application via RPC, as this is external to OTD you may close this issue. Open another one if there's any blocker for implementing such RPC.

fredemmott, 2022-03-04:
Ok, doing something specific to my app seems relatively straightforward. Closing as there does not appear to be interest here in a more general solution

fredemmott, 2023-01-28:
I eventually did this as https://github.com/OpenKneeboard/OTD-IPC