OpenTabletDriver/OpenTabletDriver #2111: Add support for old xp-pen g430 / xp-pen star 04

AnimeFreak0710, 2022-03-09:
OS: windows & manjaro linux
OTD version: latest

I'd like to be able to use my old xp-pen g430. Early revisions of this tablet have different drivers. After a lot of back and forth with xp pen support I managed to get it to work on my windows PC with the xp-pen star 04 drivers, sadly these are not available on linux though. I'm not sure if adding support for the star 04 will make my old g430 work, but I'm hopeful.
Currently OTD will not detect my old g430 and make it stop working completely, requiring a full restart to get it to work again. The tablet is still detected in my usb devices as seen in this image, it just stops working. 
[image]

jamesbt365, 2022-03-09:
the old variations should be supported just fine, check the FAQ on the otd wiki got device in use and cannot open device streams

AnimeFreak0710, 2022-03-12:
> the old variations should be supported just fine, check the FAQ on the otd wiki got device in use and cannot open device streams

I have followed all the steps on the FAQ, still doesn't work.

I have also contacted xp-pen to see if they have any compatible linux drivers available, but I got this reply: 
[image]

jamesbt365, 2022-03-12:
Well, help > export diagnostic, save that file as a `.txt` and send it here

jamesbt365, 2022-06-30:
Closing due to inactivity and no response

closed 2022-06-30 by jamesbt365

A1caida, 2022-10-03:
I have the same issue. Ubuntu 20.04
diagnostic file
[xppen430.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9699857/xppen430.txt)

jamesbt365, 2022-10-03:
You can take the configuration from #2115 and replace the one in the build you are using but it can't be in OTD natively because they have the same strings and they'll conflict.

A1caida, 2022-10-03:
> You can take the configuration from #2115 and replace the one in the build you are using but it can't be in OTD natively because they have the same strings and they'll conflict.

It works! Thank you!