OpenTabletDriver/OpenTabletDriver #2118: Add Wacom CTL-6100WL Bluetooth support!

ShadowOfPripyat, 2022-03-12:
## Description

Hello, i bought the **Wacom Intuos M** (**Bluetooth** model) a few days ago and i found out this page from the game "**osu!**", i followed the steps to install your drivers but Bluetooth doesn't work.

I think that this is an important feature because this tablet is a bit expensive and its main feature its being **wireless**

[image]
[image]

Thank You :D

IllustratedMan-code, 2023-02-02:
This is currently impossible to implement as long as opentabletdriver is dependent on HIDSharp, see #2628

ShadowDeveloper, 2024-03-08:
It is a bluetooth tablet. No support

ShadowOfPripyat, 2024-03-08:
The smaller model of this tablet is supported with bluetooth. why this bigger model isn't?

IllustratedMan-code, 2024-03-08:
Bluetooth tablets in general are not supported, see

https://github.com/OpenTabletDriver/OpenTabletDriver/issues/3047

hey2022, 2026-09-06:
I managed to get aux buttons working on linux, but the code I wrote probably ins't that great (I don't really know anything about tablet drivers):
1. Build HIDSharpCore with [#28](https://github.com/OpenTabletDriver/HIDSharpCore/pull/28) and the fix mentioned in #4672 (https://github.com/hey2022/HIDSharpCore/tree/basic-bluetooth)
2. Build OTD with #4672 and the config for CTL-6100WL mentioned by @Dutchs with another patch for aux buttons (https://github.com/hey2022/OpenTabletDriver/tree/bluetooth-ctl6100wl) and replace the `HidSharpCore.dll` with the previously patched build.
3. I also had to add udev rules for bluetooth permissions: `SUBSYSTEM=="hidraw", MODE="0666", TAG+="uaccess", TAG+="udev-acl"` or else there would be an error:
	```
	[DeviceUnauthorizedAccessException:Debug]	Not permitted to open HID class device at /dev/hidraw1.
	   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
	   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
	   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
	   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
	   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 25
	   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
	```
	I tried making the udev rule more specific, but for some reason it just wont match correctly.

For people on NixOS you can refer to my commit that got everything working (https://github.com/hey2022/dotfiles/commit/a6926e82220fcbddbe1f37612a8ecf36f4b60921)

Tsuyumi25, 2026-09-09:
@Dutchs @hey2022, thanks for the CTL-6100WL Bluetooth investigation and testing in this thread; #4672 now includes the device configuration, four ExpressKeys, report-state handling, and the generated Linux hidraw permission rule. A fresh PID `0x03C8` Tablet Debugger recording covering pen input, proximity exit/re-entry, and all four ExpressKeys would be welcome when convenient.

For an additional Linux integration check, please use https://github.com/OpenTabletDriver/HIDSharpCore/pull/31, disable any custom broad hidraw rule such as `MODE="0666"`, install the rules generated by #4672, and reconnect the tablet before launching OTD. Confirming that OTD detects it as a regular user would verify the Linux permission path too. The recording alone is sufficient for the CTL-6100WL parser behavior.