OpenTabletDriver/OpenTabletDriver #2119: Support Request: Pen detectable in tablet debugger but cursor can't move

robbiegumgum, 2022-03-12:
# Support Request
I'm using Wacom CTL-6100 wired (app shows CTL-6100WL ). On the tablets supported section I cant see CTL-6100 dunno if that's the issue.
My pen signals can be detected when touching tablet but i cant move cursor
I tried saving and applying and everything but to no avail. Here're the screenshots

[image]
[image]

Pls help

jamesbt365, 2022-03-12:
Attach a diagnostic as shown in the template instead of cutting it off. `Help > export diagnostic`, save that file with a .txt extension and send that file here.

Also make sure valorant or vanguard isn't open and file > reset to default

robbiegumgum, 2022-03-12:
> Attach a diagnostic as shown in the template instead of cutting it off. `Help > export diagnostic`, save that file with a .txt extension and send that file here.
> 
> Also make sure valorant or vanguard isn't open and file > reset to default

[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8237937/diag.txt)

Here it is

jamesbt365, 2022-04-19:
Fixed in `Configs` branch will be in a release soon, that or download an action build.

closed 2022-04-19 by jamesbt365