OpenTabletDriver/OpenTabletDriver #2127: Wacom CTE-450 Detected but cursor not moving

MPGlenn, 2022-03-16:
## Description

I'm assuming this is the same as the other threads like this where I just have to upload the diagnostic for someone to then make 
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System    Windows 10     |
| OpenTabletDriver Version 0.6.0.3 |
| Tablet        CTE-450           |
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8274707/diag.txt)

MPGlenn, 2022-03-16:
All other drivers are all uninstalled as well

jamesbt365, 2022-03-16:
>someone to then make a config file that I can use

wrong
1) Close valorant and vanguard, this includes task manager.
2) File > reset to defaults.

MPGlenn, 2022-03-17:
It's still not working.
I also tried going through every startup program and disabling unnecessary stuff and then rebooting, file > reset to defaults

jamesbt365, 2022-03-17:
If this is specific applications (should of been made clear earlier) do not run them as admin. Otherwise tablet > tablet debugger, move your pen around. Do you see a blue dot and the data change?

MPGlenn, 2022-03-17:
I see data change but no blue dot
OTD is not being run as admin either