OpenTabletDriver/OpenTabletDriver #2134: I can't open plugin manager

NorvalBlast, 2022-03-23:
| Operating System Win 10|
| OpenTabletDriver Version 0.6.0.3|
| Tablet CTL-461 |

jamesbt365, 2022-03-23:
follow the warning when opening opentabletdriver.

closed 2022-03-23 by jamesbt365

NorvalBlast, 2022-03-23:
Nothing

jamesbt365, 2022-03-23:
[image]
 looks like this.

NorvalBlast, 2022-03-23:
I think that I'm doing everything righе

jamesbt365, 2022-03-23:
What user do you log in with?

NorvalBlast, 2022-03-23:
I have 1 user on pc

jamesbt365, 2022-03-23:
That wasn't the question, i asked what username

NorvalBlast, 2022-03-23:
I badly understand english sry. Norval

jamesbt365, 2022-03-23:
What is you're UAC setting set to?

NorvalBlast, 2022-03-23:
I checked from lowest to highest, now lowest

NorvalBlast, 2022-03-23:
Lol, i checked every level. I checked it again and its working

NorvalBlast, 2022-03-23:
I'm sorry I wasted your time