OpenTabletDriver/OpenTabletDriver #2135: System.NullReferenceException

fluteze, 2022-03-24:
## Description
Daemon aborts at startup with the following error:

Unhandled exception. System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> System.NullReferenceException: Object reference not set to an instance of an object.

## System Information:
Linux <redacted> 5.16.15-201.fc35.x86_64 #1 SMP PREEMPT Thu Mar 17 05:45:13 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
OTD Version: 0.6.0.3+d18345a9

dotnet --version
6.0.201

Tablet: Gaomon PD1161

Startup log:

[<redacted>@<redacted> bin]$ ./OpenTabletDriver.Daemon 
[Plugin:Debug]	Loading plugin 'Precision Control'
[Plugin:Debug]	Loading plugin 'VMultiMode'
[Plugin:Debug]	Loading plugin 'Windows Ink'
[Detect:Info]	Searching for tablets...
[Detect:Debug]	Searching for tablet '10moon 1060N'
[Detect:Debug]	Searching for tablet 'Acepen AP 1060'
[Detect:Debug]	Searching for tablet 'Artisul M0610 Pro'
[Detect:Debug]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M106K'
[Detect:Debug]	Searching for tablet 'Gaomon M10K Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M10K'
[Detect:Debug]	Searching for tablet 'Gaomon M1220'
[Detect:Debug]	Searching for tablet 'Gaomon M1230'
[Detect:Debug]	Searching for tablet 'Gaomon PD1161'
[Device:Debug]	Initializing device 'Tablet Monitor' /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/0003:256C:006D.0001/hidraw/hidraw0
[Device:Debug]	Using report parser type 'OpenTabletDriver.Configurations.Parsers.UCLogic.UCLogicReportParser'
[Device:Debug]	Initialized string index 200
[Detect:Info]	Found tablet 'Gaomon PD1161'
[Detect:Debug]	Searching for tablet 'Gaomon PD1560'
[Detect:Debug]	Searching for tablet 'Gaomon S56K'
[Detect:Debug]	Searching for tablet 'Gaomon S620'
[Detect:Debug]	Searching for tablet 'Gaomon S630'
[Detect:Debug]	Searching for tablet 'Genius G-Pen 560'
[Detect:Debug]	Searching for tablet 'Genius i608x'
[Detect:Debug]	Searching for tablet 'Huion 420'
[Detect:Debug]	Searching for tablet 'Huion G10T'
[Detect:Debug]	Searching for tablet 'Huion GC610'
[Detect:Debug]	Searching for tablet 'Huion GT-220 V2'
[Detect:Debug]	Searching for tablet 'Huion GT-221 Pro'
[Detect:Debug]	Searching for tablet 'Huion H1060P'
[Detect:Debug]	Searching for tablet 'Huion H1161'
[Detect:Debug]	Searching for tablet 'Huion H420'
[Detect:Debug]	Searching for tablet 'Huion H420X'
[Detect:Debug]	Searching for tablet 'Huion H430P'
[Detect:Debug]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Debug]	Searching for tablet 'Huion H610 Pro'
[Detect:Debug]	Searching for tablet 'Huion H640P'
[Detect:Debug]	Searching for tablet 'Huion H950P'
[Detect:Debug]	Searching for tablet 'Huion HC16'
[Detect:Debug]	Searching for tablet 'Huion HS610'
[Detect:Debug]	Searching for tablet 'Huion HS611'
[Detect:Debug]	Searching for tablet 'Huion HS64'
[Detect:Debug]	Searching for tablet 'Huion Kamvas 13'
[Detect:Debug]	Searching for tablet 'Huion Kamvas 16 (2021)'
[Detect:Debug]	Searching for tablet 'Huion Kamvas 16'
[Detect:Debug]	Searching for tablet 'Huion Kamvas 20'
[Detect:Debug]	Searching for tablet 'Huion Kamvas 22 Plus'
[Detect:Debug]	Searching for tablet 'Huion Kamvas Pro 12'
[Detect:Debug]	Searching for tablet 'Huion Kamvas Pro 13'
[Detect:Debug]	Searching for tablet 'Huion Kamvas Pro 16'
[Detect:Debug]	Searching for tablet 'Huion Kamvas Pro 20'
[... 122 lines cut ...]
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640S'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S'
Unhandled exception. System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at VoiDPlugins.Library.VoiD.GlobalStore`1.GetOrInitialize[U](TabletReference tabletReference, Func`1 initialValue) in /home/runner/work/VoiDPlugins/VoiDPlugins/src/VoiDPlugins.Library/VoiD/GlobalStore.cs:line 21
   at VoiDPlugins.OutputMode.VMultiAbsolutePointer..ctor(TabletReference tabletReference, IVirtualScreen virtualScreen) in /home/runner/work/VoiDPlugins/VoiDPlugins/src/OutputMode/VMultiMode/Pointer/VMultiAbsolutePointer.cs:line 19
   at VoiDPlugins.OutputMode.VMultiAbsoluteMode.Initialize() in /home/runner/work/VoiDPlugins/VoiDPlugins/src/OutputMode/VMultiMode/Output/VMultiAbsoluteMode.cs:line 25
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Span`1& arguments, Signature sig, Boolean constructor, Boolean wrapExceptions)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at OpenTabletDriver.Desktop.Reflection.PluginSettingStore.TriggerEventMethods(Object obj, TabletReference tabletReference) in /usr/local/src/OpenTabletDriver/OpenTabletDriver.Desktop/Reflection/PluginSettingStore.cs:line 174
   at OpenTabletDriver.Desktop.Reflection.PluginSettingStore.Construct[T](TabletReference tabletReference, Boolean trigger) in /usr/local/src/OpenTabletDriver/OpenTabletDriver.Desktop/Reflection/PluginSettingStore.cs:line 57
   at OpenTabletDriver.Daemon.DriverDaemon.SetSettings(Settings settings) in /usr/local/src/OpenTabletDriver/OpenTabletDriver.Daemon/DriverDaemon.cs:line 163
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings() in /usr/local/src/OpenTabletDriver/OpenTabletDriver.Daemon/DriverDaemon.cs:line 222
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
Aborted (core dumped)

If you need any additional information or debug access, this can be provided.

jamesbt365, 2022-03-24:
Answers given in the stack trace, vmulti plugin (broken on 0.6 and its windows only anyway), remove it from `~/.config/OpenTabletDriver/Plugins` you may also need to delete the settings file.

closed 2022-03-24 by jamesbt365

fluteze, 2022-03-24:
Yes, it is a windows only plugin; however, it still generates an unhandled error exception on linux. It is the 'unhandled' part that I'm reporting as an error, not the existence of the problematic plugin.

jamesbt365, 2022-03-24:
The unhandled exception happens on __all__ platforms because OpenTabletDriver changed the plugin api and the plugin couldn't be fixed and will be fixed soon:tm: