OpenTabletDriver/OpenTabletDriver #2136: My Veikk S640 is detected instantly but my Veikk a15 pro does not

DrunkenMasterTim, 2022-03-25:
I have tried everything uninstalled drivers reinstalled countless times reinstalled framework, my a15 pro will not detect at all but my s640 connects instantly everytime

jamesbt365, 2022-03-25:
Help > Export diagnosis, save that file as a `.txt` and attach it here

DrunkenMasterTim, 2022-03-27:
[diagnosis.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8356821/diagnosis.txt)

jamesbt365, 2022-03-27:
```    {
      "Time": "2022-03-27T00:38:30.0235497-04:00",
      "Group": "Detect",
      "Message": "Another tablet driver found: Veikk",
      "StackTrace": null,
      "Level": 2,
      "Notification": false
    },
    {
      "Time": "2022-03-27T00:38:30.0235878-04:00",
      "Group": "Detect",
      "Message": "Detection for Veikk tablets might be impaired",
      "StackTrace": null,
      "Level": 2,
      "Notification": false
    },
```
uninstall then export diagnostic again.

DrunkenMasterTim, 2022-03-27:
[New Text Document.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8358423/New.Text.Document.txt)

DrunkenMasterTim, 2022-03-28:
[New Text Document.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8363424/New.Text.Document.txt)
use this one didnt have tablet plugged in for the last one

jamesbt365, 2022-04-02:
Create a `Configurations` folder by the executables of OpenTabletDriver and move [this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8402537/A15.Pro.json.txt) here, then remove the `.txt` at the of the file teh name leaving it as `A15 Pro.json`

DrunkenMasterTim, 2022-04-02:
i just did it, its detected but now when i hover over the tablet the cursor just goes crazy up and down

jamesbt365, 2022-04-24:
fixed in #2212

closed 2022-04-24 by jamesbt365