OpenTabletDriver/OpenTabletDriver #2139: Huion 420 Tablet Detection Problem on Latest OTD Version

Gzenn23, 2022-03-29:
## Description

Apparently on version 0.6.0.3 as well the previous 0.6.0.2 did not detect the Tablet Huion 420 properly even though I already Installed the Zadig's WinUSB preferred. but when I used the 0.6.0.1 version of if, the driver detected my tablet Properly

Somehow please fix this in the future updates
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |Windows 11
| OpenTabletDriver Version |0.6.0.3
| Tablet                   |Huion 420

jamesbt365, 2022-03-29:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/configs/OpenTabletDriver.Configurations/Configurations/Huion/420.json Download that, make a `Configurations` folder by the executable of otd, move it there.

closed 2022-03-29 by jamesbt365

Gzenn23, 2022-03-29:
Thank you for your Guidance and Support! It solved the problem