OpenTabletDriver/OpenTabletDriver #2140: Support Request: Horrible delay on linux mint xfce (nvidia)

MattRBX, 2022-03-29:
there is a horrible and slightly visible delay when using my tablet on linux to play osu
it occurs basically no matter which tablet driver i use and it's not due to framedropping or anything since im running at least 1000+ fps at 0.45ms delay
also theres less delay when using a mouse but im a tablet player so i just suck

i tried looking at the FAQ on opentabletdriver.net for linux and since my laptop has a gtx-1660ti mobile i looked at the poor performance for nvidia to solve it but i cant find "Force full composition pipeline" in my nvidia settings
so then i tried looking up how to find it, get it, disable it through terminal and nothing really seemed to work

so now im stuck with a horrible delay that i'd say is about 50-100ms

**NOTE:** im not especially good at linux but i usually get stuff working with the right guides so try and keep it simple plz ty

Linux mint version: xfce 20.3
Nvidia driver: nvidia-driver-510 proprietary driver

Diagnostic Information:
{
  "App Version": "OpenTabletDriver v0.6.0.3",
  "Build Date": "02/04/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.4.0.105",
    "VersionString": "Unix 5.4.0.105"
  },
  "Environment Variables": {
    "USER": "mathias",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0.0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/home/mathias",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/mathias/.dotnet/tools"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:056A:0303.0003/hidraw/hidraw2",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Intuos PTM",
      "SerialNumber": "",
      "FriendlyName": "Intuos PTM",
      "VendorID": 1386,
      "ProductID": 771,
      "InputReportLength": 64,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:056A:0303.0002/hidraw/hidraw1",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Intuos PTM",
      "SerialNumber": "",
      "FriendlyName": "Intuos PTM",
      "VendorID": 1386,
      "ProductID": 771,
      "InputReportLength": 10,
      "OutputReportLength": 0,
      "FeatureReportLength": 266,
      "CanOpen": true
    }
  ],
  "Console Log": [
    {
      "Time": "2022-03-29T19:08:34.6602958+02:00",
      "Group": "Detect",
      "Message": "Another tablet driver found: Wacom",
      "StackTrace": null,
      "Level": 2,
      "Notification": false
[... 1506 lines cut ...]
      "Time": "2022-03-29T19:08:34.6642983+02:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-03-29T19:08:34.6643175+02:00",
      "Group": "Plugin",
      "Message": "Loading plugin 'Circular Area'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-03-29T19:08:34.6643383+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2022-03-29:
attach the diagnostic as a file, don't just throw it into the chat.
> it occurs basically no matter which tablet driver i use
not exactly an otd issue then, sounds like a configuration issue more than anything.

X9VoiD, 2022-03-29:
This is not an OTD issue. Try asking around reddit, or other forums for solutions. Most likely solution is to disable your DE's compositor, and running osu! on fullscreen.

closed 2022-03-29 by X9VoiD