OpenTabletDriver/OpenTabletDriver #2142: Multi-Key Binding doesn't support for Microsoft Whiteboard

liuandi233, 2022-03-30:
## Description

The Multi-Key Binding works fine when I bind them to keys like alt+tab and ctrl+a, but pen binding (alt+w+1) and erase binding (alt+x) does not work in Microsoft Whiteboard. My keyboard is able to toggle them.
I actually don't know if i should open issue here or in microsoft whiteboard
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows11 Build22581
| OpenTabletDriver Version | v0.6.0.3
| Tablet                   | Wacom CTL-472
| Microsoft Whiteboard Version | 22.10314.111

jamesbt365, 2022-03-30:
Does the binding work on other places on your system?

liuandi233, 2022-03-30:
idk how to test them but i think it will work because alt+tab and ctrl+a is working fine
is there any softwares that could test this?

jamesbt365, 2022-03-30:
I'm just asking if the 3 key binding works elsewhere, if it does then its not an otd bug.

liuandi233, 2022-03-30:
yes 3 key binding is working fine

gonX, 2026-10-02:
This feels like a Microsoft Whiteboard issue, especially since you're mentioning that some key bindings work. Regardless, closing as stale.

closed 2026-10-02 by gonX