OpenTabletDriver/OpenTabletDriver #2151: Add support for XP-Pen Deco Fun S and XS

Rondosu, 2022-04-01:
when I came here to download opentabletdriver for my deco fun S,
I realized that it only available for L (the largest size)

please add the S and XS for people who don't use the largest size like me

jamesbt365, 2022-04-01:
We have the L and the XS, we are just missing the small. Fill out the template appropriately and we will get started.

jamesbt365, 2022-04-01:
[image]

closed 2022-04-01 by jamesbt365

Rondosu, 2022-04-02:
so how do I install it?
I have my tablet plugged in and opentabletdriver opened
but it does not detect

Rondosu, 2022-04-02:
> [image]

so how do I install it?
I have my Deco Fun S (CT640 I think) and opentabletdriver on
but it not detecting my Deco Fun S