OpenTabletDriver/OpenTabletDriver #2153: Windows Ink Mode Cursor Lock

SupaJuke, 2022-04-02:
## Description

This is on Window Ink mode (on 0.6.0.3). Whenever I lift my pen up and start to use a mouse instead, the mouse cursor would teleport to the last position that I lifted the pen, especially when I stop moving my mouse (as if the cursor is fixed to the position).

https://user-images.githubusercontent.com/25064349/161371430-ac3ed41c-8087-4546-a102-9c0b246be715.mp4

Here, I lifted my pen on the top right corner of the screen and then started using a mouse. As you can see, the cursor would warp back to the last position I lifted my pen.

I'm not sure if this is a bug or not, so to be safe I decided to put it here. Please change it to bug if indeed this is not a desirable behavior.

SupaJuke, 2022-04-02:
After downgrading to 0.5.3.3, the issue "no longer" appears: the jump / warp still happens when I switch from using a pen to maneuver the cursor around to mouse, but it happens only once.

With that being said, I would appreciate if a mod could change this into a bug report thread instead. Thanks!

jamesbt365, 2022-04-02:
#1124

closed 2022-04-02 by jamesbt365